wpilibsuite/sysid #14: Figure out project generation...

prateekma, 2021-02-15:
I have the Generator GUI and @Piphi5 has some generation code sitting around. We need to decide whether we should merge this or create another avenue for teams to log data (i.e. API in WPILib).

PeterJohnson, 2021-02-15:
I think there's two potential alternatives to project generation:
1) Provide API to interface with characterization for integration into the team "main" robot program.  Some specific things to figure out here have to do with eliminating potential sources of timing issues--e.g. disabling LiveWindow and any other command code, setting loop periodic period, real time thread status, etc.  The biggest downsides to this approach are that it's near impossible to eliminate *every* source of error, and it does moderately increase the barrier to entry, as teams will need to read API docs and write code to integrate it in their program.  The biggest advantage is that it eliminates sources of error between the team's robot configuration and the sysid configuration--e.g. if they have teleop working with encoders+gyro, the API to pull in that data will use exactly the configuration they've set up.  There are some smaller advantages too, like making it easier to do recharacterization.

2) Provide a canned C++ project that is data-driven.  So sysid just outputs configuration (either a file, or through NT, or both) to it and the project takes care of everything.  If we want to really be fancy, this could potentially be a mostly-static executable with all the vendor options pre-integrated so it can simply be deployed rather than user-built.  While this still requires duplication of robot configuration, it's easier to use than generating+building a custom project.

Some other broad goals to consider... with either project generation or option 2, we might be able to provide a e.g. Drivetrain base class that loads the same configuration and instantiates the right things.  This is harder to do with option 2 (compared to code gen) due to vendor dependencies (the common Drivetrain class would need to depend on all the vendor bits, not just what the user actually needs, at least for C++).

Piphi5, 2021-02-16:
So, just to clarify, option 2 would essentially be some sort of data transmitted to a generalized program that would then be able to log the proper information based on the sent config?

PeterJohnson, 2021-02-16:
Yes, option 2 would be a generalized program that configures the motors and sensors per the sent config.

Piphi5, 2021-02-16:
So does option 2 do all of its work at runtime? If so, how could we accomplish that?

PeterJohnson, 2021-02-17:
Yes.  Essentially the approach is to have an array of every type of motor controller, and “new” and configure them at runtime (so dynamic configuration) based on the config inputs.

Piphi5, 2021-02-19:
@PeterJohnson does [this](https://github.com/wpilibsuite/sysid/blob/b4ce9a6b99c0aa1f360892d92f316888ebbcfc42/base_projects/single-sided/src/main/cpp/Robot.cpp) robot code resemble what you were imagining?

PeterJohnson, 2021-02-19:
Yes, something along those lines.  We typically try to avoid dynamic casts, but if it’s not being called frequently it’s probably okay, just a bit of a code smell.

Piphi5, 2021-02-19:
What other options are there for casting the Speeds controllers into their specific types?

PeterJohnson, 2021-02-19:
Avoid the cast by doing a common interface with a wrapper class implementation for each motor controller.

prateekma, 2021-02-19:
I think the biggest reason for the casts is to access encoders that are connected to the motor controller itself. The `SpeedController` interface is sufficient for doing everything related to a motor controller.

We could add a separate `Encoder` interface that takes references to the smart controllers (note this will still involve a dynamic_cast but only once) or simply assume all the casts will be safe (because we're checking the JSON value) and using `static_cast` instead.

Piphi5, 2021-02-20:
Do note that the `static_cast` approach would still require the `dynamic_cast` to be used for CTRE motor controllers as they use virtual inheritance, but the number of dynamic casts are still significantly reduced.

Piphi5, 2021-02-21:
@PeterJohnson should we switch to the wrapper class setup or stick to the [static/dynamic cast setup ](https://github.com/wpilibsuite/sysid/blob/44df33aa50ed6f50b655dc624e59201e84a9c3d3/base_projects/single-sided/src/main/cpp/Robot.cpp)?

PR #24 (2021-04-26): Add initial project generation infrastructure
TODO:
Fixes #14 
- [x] Finalize Generator GUI
- [x] Finish Single-Sided Motor Project
- [x] Finish Drive Project
- [x] Create Tests to ensure the validity of these projects
- [x] Fix #37 By adding Romi Preset
- [x] Update data to match #38 
- [x] Add the proper CPR conversions for motor controllers
- [x] Fix #41 by adding SysId Classes
- [x] Fix widget sizing
- [x] ~~Set CAN update rates to 5 ms (need to figure out CAN issues)~~ (File an issue, and resolve later so that we can get the fundamental generator merged + tested sooner)
- [x] Have setting [... cut]
closed 2021-04-26 by prateekma