wpilibsuite/sysid #20: Change format of data collection

prateekma, 2021-02-17:
The current format wastes extra space for mechanisms other than drivetrain and sends some unnecessary data. We should change the format as early as possible.

A small view for converting a JSON from the old format to the new one would also be nice.

Piphi5, 2021-02-17:
Would a CSV-esque telemetry format (Commas separated by newlines) be best?.

Then for drivetrain, we tack on the additional values to the end?

prateekma, 2021-02-17:
I was thinking the following for mechanisms other than linear drivetrain (simple, elevator, arm, angular drivetrain):
```
timestamp, motor voltage, position, velocity
```
Note that `position` and `velocity` would be gyro angle and gyro rate for the angular drivetrain test.

For linear drivetrain:
```
timestamp, l voltage, r voltage, l position, r position, l velocity, r velocity
```

We would then basically send the data as we do now: collect it during auto and send a giant blob after the robot is disabled.

@calcmogul Thoughts?

calcmogul, 2021-02-17:
Are we ok with flywheels not using `position`? I guess it's fine since position is at least defined in that case. We have the same dilemma for drivetrains though.

For linear vs angular drivetrain, it may be inconvenient to change the robot code between linear and angular tests. If we added angle and angular rate to the measurement set so we have `time, l voltage, r voltage, l position, r position, l velocity, r velocity, angle, angular rate`, we could switch between the tests in the GUI. The data would be defined in both test types, but most of it wouldn't be used in the angular case.

prateekma, 2021-02-17:
Well, for linear and angular they would have to be separate tests (JSONs) anyway because one would involve the robot moving back and forth and the other involves spinning. So the angle data for the linear JSON would be thrown out completely and vice versa with the encoder data for the angular tests.

The exception here is track width, where we need both, so I would be okay with both drivetrain tests using what you described.

calcmogul, 2021-02-17:
Yeah, they'd be separate JSONs. My point is we'd want the same robot program to work for all tests related to a given subsystem. They'd change code to characterize a different subsystem.

calcmogul, 2021-02-17:
If we did have the client only send what's needed, there'd be:
velocity system:
```
timestamp, motor voltage, velocity
```
position system:
```
timestamp, motor voltage, position, velocity
```
linear drivetrain:
```
timestamp, l voltage, r voltage, l position, r position, l velocity, r velocity
```
angular drivetrain:
```
timestamp, l voltage, r voltage, l position, r position, l velocity, r velocity, angle, angular rate
```

prateekma, 2021-02-20:
I think I will move forward with the following formats.

Drivetrain (Linear, Angular):
```
timestamp, l voltage, r voltage, l position, r position, l velocity, r velocity, angle, angular rate
```

Other Mechanisms (Simple, Elevator, Arm):
```
timestamp, voltage, position, velocity
```

We don't need the position data for a flywheel, but it's useful for other mechanisms that might use the "Simple" mode (e.g. a swerve module). The position data is only used for letting the user know how far their mechanism traveled.

Piphi5, 2021-02-20:
Also, this matches the 2 different types (Single-sided, Drivetrain) of projects that would be deployed on the Rio.

closed 2021-03-07 by prateekma