wpilibsuite/sysid #37: Add Romi Preset

Daltz333, 2021-02-26:
frc-char requires special setup for characterizing a Romi. While constants are mostly plug-n-play for users of Romi, it's a great learning platform, and we should add a preset for Romi.

Piphi5, 2021-02-26:
Would this just be for the Romi drivetrain?

Daltz333, 2021-02-26:
Yes.

Piphi5, 2021-02-26:
Are the ports, epr, etc. all fixed? Or should they be able to change that in the GUI?

Daltz333, 2021-02-26:
It's all fixed. The Romi project is simulation with the websockets enabled and pointed toward 10.0.0.2

Piphi5, 2021-02-26:
Alright, I think we're going to add an option in #24 that'll generate a fixed config file for it.

Piphi5, 2021-03-01:
@Daltz333 Would you be able to test #24 on your Romi once it's finalized?

Daltz333, 2021-03-01:
Yes

Piphi5, 2021-03-03:
@Daltz333 Could you test PR #24 on a Romi?

You would probably need to do the following procedure:
1. Pull the branch
2. Run the GUI
3. Use the `Romi` preset on the generator and save the config
4. Deploy the Drivetrain project (which will be located in the `base_projects/drivetrain` folder)
5. Run telemetry and analysis and see if the numbers roughly match up with the old tool.

Daltz333, 2021-03-03:
Can you send me a compiled executable?

Piphi5, 2021-03-03:
Actually, I'll have to send everything after #38 is merged. 

You would probably still need to pull the files as of right now just because we haven't fully figured out how we're going to deploy the project so the GUI saves the config into a hardcoded subdirectory in the repo.

prateekma, 2021-03-03:
I'll try to get 38 merged by Thursday night

Piphi5, 2021-03-03:
Alright, no rush. I still have some changes to make as well.

PR #24 (2021-04-26): Add initial project generation infrastructure
TODO:
Fixes #14 
- [x] Finalize Generator GUI
- [x] Finish Single-Sided Motor Project
- [x] Finish Drive Project
- [x] Create Tests to ensure the validity of these projects
- [x] Fix #37 By adding Romi Preset
- [x] Update data to match #38 
- [x] Add the proper CPR conversions for motor controllers
- [x] Fix #41 by adding SysId Classes
- [x] Fix widget sizing
- [x] ~~Set CAN update rates to 5 ms (need to figure out CAN issues)~~ (File an issue, and resolve later so that we can get the fundamental generator merged + tested sooner)
- [x] Have setting [... cut]
closed 2021-04-26 by prateekma