wpilibsuite/sysid #41: Refactor NT logging in integration_test_project into SysIdLogger classes

calcmogul, 2021-03-04:
Even if we do code generation, the abstraction there will be helpful in codegenned projects as well as teams who decide to gather characterization data in situ anyway.

prateekma, 2021-03-04:
We should also discuss whether we want to add these logger classes to WPILib so that teams can use it from their regular robot code.

Piphi5, 2021-03-04:
Should we have it as an interface where they just implement methods to collect data and set motor speed or should it be an abstract class that also handles the Network Tables reading and string publishing?

calcmogul, 2021-03-04:
It should handle the reading and publishing stuff that's in https://github.com/wpilibsuite/sysid/blob/main/integration_test_project/src/main/cpp/Robot.cpp. SimpleMotor would have the following functions:

```cpp
class SimpleMotorLogger {
  SimpleMotorLogger() {
    // Disable LiveWindow telemetry
    // Create necessary NT entries with SmartDashboard put functions
  }

  // Called by user in AutonomousPeriodic() or TestPeriodic()
  void Log(units::volt_t batteryVoltage, units::volt_t motorVoltage, double velocity) {
    // Get timestamp here with frc2::Timer::GetFPGATimestamp()
    // Fill in data array and log it
  }

  // Called by user in DisabledInit()
  void SendData() {
    // Sends the giant data buffer over NT
  }
};
```
Their contents would roughly correspond to what's in AutonomousPeriodic() and DisabledInit() in the previous Robot.cpp.

Piphi5, 2021-03-05:
Should any changes be made to [these logging classes](https://github.com/Piphi5/sysid/tree/generation/integration_test_project/src/main/include/logging)?

calcmogul, 2021-03-05:
The APIs look good, but the implementations look incomplete.

Piphi5, 2021-03-05:
Implementations as in integrating it into the respective robot projects?

calcmogul, 2021-03-05:
No. The classes themselves don't appear to be finished.

calcmogul, 2021-03-05:
Oh, some stuff was hiding in the base class.

calcmogul, 2021-03-05:
I have some other comments for cleanup, but I'll save those for a PR.

PR #24 (2021-04-26): Add initial project generation infrastructure
TODO:
Fixes #14 
- [x] Finalize Generator GUI
- [x] Finish Single-Sided Motor Project
- [x] Finish Drive Project
- [x] Create Tests to ensure the validity of these projects
- [x] Fix #37 By adding Romi Preset
- [x] Update data to match #38 
- [x] Add the proper CPR conversions for motor controllers
- [x] Fix #41 by adding SysId Classes
- [x] Fix widget sizing
- [x] ~~Set CAN update rates to 5 ms (need to figure out CAN issues)~~ (File an issue, and resolve later so that we can get the fundamental generator merged + tested sooner)
- [x] Have setting [... cut]
closed 2021-04-26 by prateekma