wpilibsuite/sysid #47: Integration tests only uses smartdashboard to change mechanism

KyleQ1, 2021-03-04:
When running the integration tests it seemed to only change mechanism when we changed the value from Drivetrain on SmartDashboard instead of using what's in sysid.

calcmogul, 2021-03-04:
You mean the integration test project, right?

prateekma, 2021-03-04:
It was really intended as an internal thing but we can publish the mechanism to NT in TelemetryManager as well

calcmogul, 2021-03-04:
We mainly did this because frc-char was unreliable and we needed working telemetry. Robot-side telemetry logging classes would mean we no longer have to use the integration test project.

Piphi5, 2021-03-04:
We could probably have the Telemetry and Logging classes finalized by maybe next week? Then you could probably paste the files into your code.

calcmogul, 2021-03-04:
We got good results with this approach and don't need to characterize anything else atm, so no rush.

closed 2021-03-07 by prateekma