wpilibsuite/sysid #108: Integration Test Projects that don't build don't terminate

Piphi5, 2021-04-19:
As seen [here](https://github.com/Piphi5/sysid/runs/2385236947?check_suite_focus=true), whenever the integration test project fails to build, the test waits indefinitely to connect to the robot. Either terminating the test whenever the status for `std::system` != 0 or making the connect wait have a time out would fix this.

calcmogul, 2021-04-19:
We already assert if `std::system()` returns nonzero, so `std::system()` probably isn't exiting.
https://github.com/wpilibsuite/sysid/blob/main/src/integrationtest/native/cpp/SimulationTest.cpp#L70-L71

Piphi5, 2021-04-19:
Looks like the assert happens but it still continues the test:
```
BUILD FAILED in 7s
D:\a\sysid\sysid\src\integrationtest\native\cpp\SimulationTest.cpp(65): error: Expected equality of these values:
  0
  result
    Which is: 1
The robot program couldn't be started
[ RUN      ] IntegrationTest.Drivetrain
Running: slow-forward
```