wpilibsuite/sysid #116: CAN Status Frame Periods

Piphi5, 2021-04-26:
Currently, CAN Status Frame Periods are set to the defaults which means that sensor readings are updated at 10ms or 20ms (depending on the smart motor controller) which doesn't line up with the project's 5ms loop period. 

However, there seems to be issues with lowering these periods which causes the sensor readings to randomly be reset (probably need to figure out issues with the vendors).

calcmogul, 2021-04-26:
The sensor readings resetting may be caused by CAN bus saturation and packet collisions. At least, that's what I saw when trying to set the SPARK MAX encoder position status frame to 10ms.

calcmogul, 2021-05-26:
If increasing the status frame periods for unnecessary data isn't enough to fix it, there's nothing else we can do. FRC's CAN spec doesn't let us manipulate the message IDs to specify message priority.

calcmogul, 2021-10-15:
I'm closing this since we don't plan on pursuing this further. If someone wants to revisit this, they can.

closed 2021-10-15 by calcmogul