wpilibsuite/sysid #146: Weird end behavior for some of the simulation graphs

Piphi5, 2021-05-27:
I loaded the data from #74 and it looks like the simulation is quite off towards the end:
[image]

calcmogul, 2021-05-27:
That's because Ka < 0.

Piphi5, 2021-05-27:
Yeah, somethings messing with that kA fit since its clearly >0

calcmogul, 2021-06-03:
I thought the broken fit was caused by the velocity spikes at the beginning of the quasistatic tests.
[image]

This printed removals, but didn't seem to help:
```cpp
  // Remove data with velocity spikes
  sysid::ApplyToData(
      preparedData,
      [&](std::string_view key) {
        auto& data = preparedData[key];

        data.erase(std::remove_if(
                       data.begin(), data.end(),
                       [=](const auto& pt) {
                         if (std::abs(pt.velocity - pt.nextVelocity) > 25.0) {
                           fmt::print(stderr, "{}: FOUND with {}\n", key,
                                      std::abs(pt.velocity - pt.nextVelocity));
                         }
                         return std::abs(pt.velocity - pt.nextVelocity) > 25.0;
                       }),
                   data.end());
      },
      [](std::string_view key) {
        return key.find("raw") == std::string_view::npos;
      });
```

Piphi5, 2021-06-03:
That is incredibly weird... Would it be safe to assume that data wouldn't get that messy? Also, why doesn't the median filter reject the outlier?

calcmogul, 2021-06-03:
I think something's wrong with the data collection itself.

Piphi5, 2021-06-03:
Probably with how the user set it up yeah

Piphi5, 2021-06-09:
This seems to be more of an issue with the collected data from the frc-char tool than an issue with sysid.

closed 2021-06-09 by Piphi5