wpilibsuite/sysid #161: SysId is Segfaulting

Piphi5, 2021-06-07:
Currently, main is segfaulting whenever there is a sysid.ini file present.

The last commit that doesn't fail is this one: https://github.com/wpilibsuite/sysid/commit/34fefdb837105c4bb379a0c4d7cfaae030a3e846.

This looks like it may be due to some bug in the project deploying infrastructure.

calcmogul, 2021-06-07:
Could you attach the sysid.ini to compare against someone who doesn't have the issue?

Piphi5, 2021-06-07:
[sysid.ini.txt](https://github.com/wpilibsuite/sysid/files/6611806/sysid.ini.txt)

calcmogul, 2021-06-07:
Here's mine, which also exhibits the issue.
[sysid.ini.txt](https://github.com/wpilibsuite/sysid/files/6611802/sysid.ini.txt)

calcmogul, 2021-06-07:
GDB output:
```
Starting program: /home/tav/git/wpilib/sysid/sysid-application/build/install/sysid/debug/lib/sysid 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[New Thread 0x7ffff7618640 (LWP 165573)]
[New Thread 0x7ffff6e17640 (LWP 165574)]
[New Thread 0x7ffff6616640 (LWP 165575)]
[New Thread 0x7ffff5e15640 (LWP 165576)]

Thread 1 "sysid" received signal SIGSEGV, Segmentation fault.
0x00005555567b0387 in wpi::detail::ConsumeSignedInteger (str="", radix=1472501520, 
    result=@0x7fffffffdab0: 1)
    at /__w/allwpilib/allwpilib/wpiutil/src/main/native/cpp/StringExtras.cpp:270
270	/__w/allwpilib/allwpilib/wpiutil/src/main/native/cpp/StringExtras.cpp: No such file or directory.
(gdb) bt
#0  0x00005555567b0387 in wpi::detail::ConsumeSignedInteger (str="", radix=1472501520, result=@0x7fffffffdab0: 1) at /__w/allwpilib/allwpilib/wpiutil/src/main/native/cpp/StringExtras.cpp:270
#1  0x00005555567b0522 in wpi::detail::GetAsSignedInteger (str="", radix=1472501520, result=@0x7fffffffdab0: 1) at /__w/allwpilib/allwpilib/wpiutil/src/main/native/cpp/StringExtras.cpp:310
#2  0x000055555658e1ab in wpi::parse_integer<int, true> (str="", radix=1472501520)
    at /home/tav/.gradle/caches/transforms-2/files-2.1/49889ed33fa505237c1c4cdea4d5be4d/wpiutil-cpp-2022.0.0-alpha-1-101-g8d054c9-headers/wpi/StringExtras.h:587
#3  0x000055555684e555 in ConvertInt (value=0x555557c496f0) at /__w/allwpilib/allwpilib/glass/src/lib/native/cpp/Context.cpp:26
#4  0x000055555684ead7 in GlassStorageReadLine (entry=0x555557c49100, line=0x555557dbd092 "mode=0") at /__w/allwpilib/allwpilib/glass/src/lib/native/cpp/Context.cpp:94
#5  0x0000555556917d8b in ImGui::LoadIniSettingsFromMemory (
    ini_data=0x555557dbcae0 "[Window][Debug##Default]\nPos=60,60\nSize=400,400\nCollapsed=0\n\n[Window][###Analyzer]\nPos=375,26\nSize=552,552\nCollapsed=0\n\n[Window][Diagnostic Plots]\nPos=895,25\nSize=375,660\nCollapsed=0\n\n[Window][###Gene"..., ini_size=848) at /__w/thirdparty-imgui/thirdparty-imgui/imgui/imgui.cpp:10391
#6  0x0000555556917a7e in ImGui::LoadIniSettingsFromDisk (ini_filename=0x555557c4e3b0 "/home/tav/.config/sysid.ini") at /__w/thirdparty-imgui/thirdparty-imgui/imgui/imgui.cpp:10330
#7  0x0000555556860835 in wpi::gui::Initialize (title=0x555556e4f00a "System Identification", width=1280, height=720) at /__w/allwpilib/allwpilib/wpigui/src/main/native/cpp/wpigui.cpp:157
#8  0x00005555565833f2 in Application () at /home/tav/git/wpilib/sysid/sysid-application/src/main/native/cpp/App.cpp:199
#9  0x0000555556586e4a in main () at /home/tav/git/wpilib/sysid/sysid-application/src/main/native/cpp/Main.cpp:16
```

PeterJohnson, 2021-06-08:
There's something else going on here... radix is garbage at the wpi::parse_integer entry point, which should be impossible, as the calling code passes 10.

Piphi5, 2021-06-08:
I removed these following lines from the sysid.ini file and the program doesn't crash.
```

[GlassStorage][NetworkTables Settings]
mode=1
iniName=networktables.ini
serverTeam=localhost
listenAddress=
dsClient=1

[SysId][Analyzer]
name=
visible=1
enabled=1

[SysId][Generator]
name=
visible=1
enabled=1

[SysId][Logger]
name=
visible=1
enabled=1

[SysId][Program Log]
name=
visible=1
enabled=1
```

PeterJohnson, 2021-06-08:
It looks like this is due to our CI using GCC 7.4 and GCC 7 having a bug that miscompiles the code in question, clobbering the stack.  See around line 42 (rsp subtraction) in https://godbolt.org/z/bhdq3YPzs

Fixing this will require moving to a more recent GCC version/updating to a newer ubuntu base version.

Piphi5, 2021-06-08:
Yes, I'm using WSL

closed 2021-06-08 by PeterJohnson

calcmogul, 2021-06-08:
I've confirmed sysid no longer crashes using the allwpilib dev artifacts built with the newer docker container.