wpilibsuite/sysid #174: RoboRIO project fails to run

willtoth, 2021-08-21:
Running the latest main build #169 on Windows x86_64, I can successfully deploy, but running the robot code gives the following error:

```
﻿﻿﻿﻿﻿﻿﻿  ﻿
﻿﻿﻿﻿﻿﻿ ********** Robot program starting ********** ﻿
﻿﻿﻿﻿﻿﻿ reading json  ﻿
﻿﻿﻿﻿﻿﻿ Project failed: [json.exception.type_error.304] cannot use at() with array ﻿
﻿﻿﻿﻿﻿﻿  ﻿
﻿﻿﻿﻿﻿﻿ ********** Robot program starting ********** ﻿
﻿﻿﻿﻿﻿﻿ reading json  ﻿
﻿﻿﻿﻿﻿﻿ Project failed: [json.exception.type_error.304] cannot use at() with array ﻿
﻿﻿﻿﻿﻿﻿ NT: server: client CONNECTED: 169.254.92.114 port 54101 ﻿
﻿﻿﻿﻿﻿﻿ NT: server: client CONNECTED: 169.254.92.114 port 54102 ﻿

```

Here is the config file (on my machine):

```json
{
  "counts per rotation": 1.0,
  "encoder type": "Built-In",
  "encoding": false,
  "gearing": 18.0,
  "gyro": "NavX",
  "gyro ctor": "SerialPort.kUSB",
  "is drivetrain": false,
  "motor controllers": [
    "SPARK MAX (Brushless)"
  ],
  "number of samples per average": 1,
  "primary encoder inverted": false,
  "primary encoder ports": [
    0,
    1
  ],
  "primary motor ports": [
    5
  ],
  "primary motors inverted": [
    false
  ],
  "secondary encoder inverted": false,
  "secondary encoder ports": [
    2,
    3
  ],
  "secondary motor ports": [
    2
  ],
  "secondary motors inverted": [
    false
  ],
  "velocity measurement period": 100
}
```

Here is the json file on the roboRIO

```json
[{"counts per rotation":1.0,"encoder type":"Built-In","encoding":false,"gearing":18.0,"gyro":"NavX","gyro ctor":"SerialPort.kUSB","is drivetrain":false,"motor controllers":["SPARK MAX (Brushless)"],"number of samples per average":1,"primary encoder inverted":false,"primary encoder ports":[0,1],"primary motor ports":[5],"primary motors inverted":[false],"secondary encoder inverted":false,"secondary encoder ports":[2,3],"secondary motor ports":[2],"secondary motors inverted":[false],"velocity measurement period":100}]
```

willtoth, 2021-08-22:
Manually removing the `[ ]` surrounding the whole json string from the above on the roboRIO appears to fix the issue.

Piphi5, 2021-08-22:
Yeah it seems like the brackets makes the program think the JSON is an array which causes the crash. Not sure why that's happening though.

Piphi5, 2021-08-24:
@willtoth can you test if the executable on #176 doesn't run into this issue?

PR #176 (2021-08-25): Fix deployed JSON generation
The initializer list constructor of the json library causes config jsons
to be converted into an array type causing issues when reading it on the
rio. Tests have been added to ensure this won't happen.

Fixes #174