wpilibsuite/sysid #175: Program crashes when loading data file

willtoth, 2021-08-22:
Attached the data file. This was saved after doing all 4 tests. 

[sysid_data20210822-104822.zip](https://github.com/wpilibsuite/sysid/files/7027725/sysid_data20210822-104822.zip)

I don't see any kind of log or other information that I can include here, is there a way to generate debugging info?

Piphi5, 2021-08-22:
Running the repository with gradle opens up a debug version and there are python plotting scripts of the data. 

I went ahead and ran the debug version and it didn't crash when opening your data. However, the graphs are definitely suspect of something going very wrong.
[image]
[image]
[image]
[image]
[image]

If you used the previous project from #174 to try to collect drivetrain data, that might be the cause of these messed up graphs.

willtoth, 2021-08-22:
Yeah this is for a non-drivetrain, so that is probably an issue. I'll try from source using gradle.

Piphi5, 2021-08-24:
@willtoth have you gotten a chance to see if not using the drivetrain characterization fixed the issue?

willtoth, 2021-08-24:
Trying to set up a better testing location, I may have some time today.

Either way, invalid input shouldn't crash the program.

Piphi5, 2021-08-24:
Alright sounds good. 

That's just to make sure there isn't a larger issue at play with data analysis.

I'll try to reproduce the issue, because exceptions should be getting caught by the analyzer.

Piphi5, 2021-08-25:
@willtoth does the program straight up crash for you or does it just throw an exception?

I'm currently unable to reproduce the issue with the data.

willtoth, 2021-08-25:
Straight up crash. Even running from cmd doesn't show any output. I'll try again with the latest.

Piphi5, 2021-08-25:
I managed to reproduce the issue with the release executable from the commit before the latest one on main. Perhaps the latest version does fix the issue.

willtoth, 2021-08-25:
Just tried #176, this one still crashes.

However I did notice that the graph loads totally fine if I launch the application from Windows cmd.

Piphi5, 2021-08-25:
Could you try running the executable from this [commit](https://github.com/wpilibsuite/sysid/runs/3411551646)?

I don't think #176 is rebased onto the latest commit yet.

Piphi5, 2021-08-28:
@willtoth did you get a chance to try the latest version on main?

Piphi5, 2021-09-13:
@willtoth PR #180 seems to fix the issue for me. Could you see if it also works on your end?

willtoth, 2021-09-15:
Works for me, thanks!

PR #180 (2021-09-15): Fix Step Voltage default data trim
Fixes #175
closed 2021-09-15 by willtoth