wpilibsuite/sysid #192: Add Romi Deploy

Piphi5, 2021-10-15:
Currently there's no way to deploy the Romi code.

veysby, 2022-02-02:
I really enjoy this tool compared to frc-characterization. Is there are any way to bypass this limitation (e.g. using https://github.com/bb-frc-workshops/romi-examples/tree/main/romi-characterization)?

veysby, 2022-02-02:
> I really enjoy this tool compared to frc-characterization. Is there are any way to bypass this limitation (e.g. using https://github.com/bb-frc-workshops/romi-examples/tree/main/romi-characterization)?

To anyone who is interested in this - I did a quick experiment and was able to connect Logger but Romi didn't respond to any test. Later I found that there are some changes in data-collection (see #131), so most likely the answer to my own question is **no** (unless if you would like to understand all the changes that was done and re-implement them)

sickingp, 2022-02-02:
This is really a shame. The main reason I purchased a Romi was to give our students an easy way to learn about characterization and trajectory following commands. When I discovered a few days ago that sysid does not support Romi, I too thought about working around this dropped functionality by using a modified version of [https://github.com/bb-frc-workshops/romi-examples/tree/main/romi-characterization](url) but it sounds like that won't work.  Is there any prospect for sysid to support Romi in the foreseeable future? Thanks.

calcmogul, 2022-02-02:
None of the existing contributors are currently working on that. Modifying the romi-characterization repo to use the new NT data format would be sufficient.

sickingp, 2022-02-03:
Thanks, Tyler.

Piphi5, 2022-02-03:
If you'd like you can try this:
1. Cloning this repo
2. Putting the attached config file in `sysid-projects/deploy`
3. Running `./gradlew :sysid-projects:drive:simulateNativeRelease -Pintegration &`  to launch the simulation GUI of the Romi project
4. 
Not sure if this will work but it might be able to run the sysid code for the Romi

veysby, 2022-02-07:
Thanks everyone for suggestions. I've tried all of them but was able to make progress only in porting changes back to Java. I'm still trying to figure out how to do the Quasistatic analysis right and would really appreciate any comments that you may have in this PR -> https://github.com/bb-frc-workshops/romi-examples/pull/15. Thank you!

calcmogul, 2023-09-21:
OBE by #518.

closed 2023-09-21 by calcmogul