wpilibsuite/sysid #199: How to use this software

mendax1234, 2021-11-04:
[image]

After clicking , what should l click and where should l download it?

calcmogul, 2021-11-04:
2021 and 2022 get installed to separate folders, so you can have them both installed at once.

Piphi5, 2021-11-04:
https://github.com/wpilibsuite/sysid/actions/runs/1414619584

If you scroll to the bottom you can see executables for your operating system that you can download and run.

calcmogul, 2021-11-04:
Interesting. The Linux binary is 207M, but all but 20M of that is symbols.

mendax1234, 2021-11-06:
Thanks，l have installed this. However ，l still don't know how to use it. Is there any instructions?

calcmogul, 2021-11-06:
These docs were updated for SysId: https://docs.wpilib.org/en/latest/docs/software/pathplanning/robot-characterization/

mendax1234, 2021-11-06:
Thanks, l will dig into it.

mendax1234, 2021-11-06:
Besides, it the feedforward different from the feedforward in CTRE? Otherwise , how can l convert feedforward in WPILIB to position control in CTRE's talonSRX

calcmogul, 2021-11-06:
You'll have to read CTRE's docs for what units they use. Maybe it's here somewhere? https://docs.ctre-phoenix.com/en/stable/

closed 2021-11-16 by calcmogul