wpilibsuite/sysid #206: Fractional decimal inputs

TateStaples, 2021-11-12:
Just wondering if it would be possible to allow for fraction inputs (ie 1/20) into double fields. Not super important feature, but might be convenient for things like Gear ratios

calcmogul, 2021-11-12:
I'm not really sure how to do this. `ImGui::InputDouble()` formats its input based on scanf-like format strings. For example, `"%.4f"` interprets the input as a floating-point number and truncates it to 4 decimal places.

Some cursory googling leads me to believe scanf() can't support this.
https://stackoverflow.com/questions/13574711/getting-a-fractional-number-with-scanf
https://stackoverflow.com/questions/15332212/reading-fractions-in-c

calcmogul, 2021-11-23:
Maybe we could write our own input function like `InputFraction()` that validates input of the form `\d(/\d)?` where `\d` means `-?[0-9]+(\.[0-9]+)?`.

Piphi5, 2021-11-25:
Is there an easy way to do regex in c++?

calcmogul, 2021-11-25:
https://en.cppreference.com/w/cpp/regext

Piphi5, 2021-11-26:
Does that mean we would have to use `InputText` and then have our own form of input validation?

calcmogul, 2021-11-26:
Yea. If you follow the call tree far enough, you'll see all the functions just use InputText().

https://github.com/ocornut/imgui/blob/master/imgui_widgets.cpp#L3568

Piphi5, 2021-11-27:
Should we have it so when they press enter on the GUI it gets converted into a decimal form or only when it gets saved to the JSON?

calcmogul, 2021-11-27:
Either of those options means it'll load with a decimal number instead of the ratio. Maybe we could have two separate input boxes with a `:` between them for gear ratio.

Piphi5, 2021-11-27:
True that'll probably be more intuitive

closed 2021-12-04 by calcmogul