wpilibsuite/sysid #209: CTRE Tachometer velocites are too large by a factor of 1024

calcmogul, 2021-11-13:
The fix may be to divide by 1024 for tachometers specifically in the following code:
```cpp
static void SetupCTREEncoder(frc::MotorController* controller,
                             FeedbackDevice feedbackDevice, int period,
                             double cpr, int numSamples, bool
                             encoderInverted, std::function<double()>&
                             position, std::function<double()>& rate) {
  auto* talonController = dynamic_cast<WPI_BaseMotorController*>(controller);
  talonController->ConfigSelectedFeedbackSensor(feedbackDevice);
  talonController->SetSensorPhase(encoderInverted);
  talonController->ConfigVelocityMeasurementWindow(numSamples);

  // Determine velocity measurement period
  auto talonPeriod = getCTREVelocityPeriod(period);

  talonController->ConfigVelocityMeasurementPeriod(talonPeriod);
  position = [=] { return talonController->GetSelectedSensorPosition() / cpr;
  }; rate = [=] {
    return talonController->GetSelectedSensorVelocity() / cpr /
           0.1;  // Conversion factor from 100 ms to seconds
  };
}
```

Piphi5, 2021-11-13:
Why doesn't setting the cpr to 1024 work?

calcmogul, 2021-11-13:
According to the person on the FRC Discord.
> But I can't cheat it by changing the CPR, that somehow is tied into the ka and kv calculations.  Possibly could make a gearing value work, but hadn't tried.  It seems like multiplying the ka and kv by 1000 is in the ballpark.  Though I don't have a firm idea of how well characterized the system is, so there is some range there.

ngreen-gt, 2021-11-13:
So setting CPR doesn't working but setting gearing to 1024 does work I've found. 

Here is that result:  
[image]

The CPR set to 1000 looked like this:
[image]

Piphi5, 2021-11-13:
Not sure why that makes a difference. The gearing multiplies the CPR:
```cpp
SetupEncoders(encoderType, isEncoding, period, cpr * gearing, numSamples,
                  controllerNames[0], m_leftControllers.at(0).get(),
                  leftEncoderInverted, leftEncoderPorts,  // m_leftCancoder,
                  m_leftEncoder, m_leftPosition, m_leftRate);
SetupEncoders(encoderType, isEncoding, period, cpr * gearing, numSamples,
                  controllerNames[0], m_rightControllers.at(0).get(),
                  rightEncoderInverted, rightEncoderPorts,  // m_rightCancoder,
                  m_rightEncoder, m_rightPosition, m_rightRate);
```

Could the hardware or something else have just been faulty when the CPR was set to 1000 or does it consistently produce that type of bad data?

Piphi5, 2021-11-20:
@ngreen-gt any updates on this issue?

ngreen-gt, 2021-11-20:
I had an issue re-running the tests, which is why initially I had problems setting CPR.  But I re-did everything and was able to set either CPR or Gearing, there was no different in the result like you noted.  

My issue I believe was possibly some combination of running the test too long (and then trying to run it again after the warning) or possibly a battery issue.  But the one test had a problem at the end my first run.  Just refreshing and re-running didn't have those problems (so I lean towards it not being battery related).  I hadn't tried to repeat allowing it to run past the warning of too much time/data logged, but may be worth trying to see if can pinpoint if that was an issue, and if I do I'll created another issue.

As far as whether the units should be fixed here, I'm not sure if there are other potential issues in doing that or not.  That may be a good solution because it isn't intuitive to do.

closed 2021-12-24 by calcmogul