wpilibsuite/sysid #215: Motion threshold is too high and isn't customizable upon JSON load failure

calcmogul, 2021-11-17:
From https://usfirst.collab.net/sf/go/post7071:

> After running the quasistatic and dynamic tests in sysID and then running the analyzer, I get the error message " Quasistatic test trimming removed all data." It is somewhat difficult to diagnose the problem without access to the project generated by sysID and deployed to the robot. Reported encoder values looked fine and the robot was driving forward and backwards correctly. I have attached the data collected from the tests here:
[sysid_data20211116-202237.json.txt](https://github.com/wpilibsuite/sysid/files/7558287/sysid_data20211116-202237.json.txt)

After plotting the data with https://github.com/wpilibsuite/sysid/blob/main/scripts/time_plots.py, it appears the data is below the default motion threshold of 0.2 (the data is around 0.04). We should probably add a more integrated way to view the raw time domain data.

Also, the JSON loading is aborted, which means the motion threshold can't be fixed by the user if there's a failure. We should fix that.

Piphi5, 2021-11-20:
Would a good solution be to disable the filtered graphs + gain calculations if it fails instead of crashing everything?

calcmogul, 2021-11-20:
Yea.

PR #220 (2021-11-27): Display raw data and let users adjust variables if filtering fails
Fixes #215 

Also removes and filtering of the raw data.
closed 2021-11-27 by calcmogul