wpilibsuite/sysid #218: sysid deploy seeks missing library

bovlb, 2021-11-21:
I've been trying to get the new sysid tool working and it seems to be dead in the water.  It claims to deploy successfully, but then there is no networktables and the DriverStation reports No Robot Code.  Looking in `FRC_UserProgram.log`, I find the error:
```error while loading shared libraries: libFRC_NetworkCommunication.so.22: cannot open shared object file: No such file or directory```
Looking on the roboRIO, only `libFRC_NetworkCommunication.so.20` is available.  Is it possible that the beta software is somehow dependent on an as-yet unreleased version of the FRC Game Tools?  Is there a workaround for this?

calcmogul, 2021-11-21:
Are you using a roboRIO imaged for 2020 or 2021? sysid was upgraded to 2022 libraries recently, which means it will only work for beta teams.

bovlb, 2021-11-21:
The roboRIO is imaged for 2021, latest available.  We're a beta team, but we're still waiting for anything to appear on TeamForge.  Once the Beta opens, will we be able to download new FRC Game Tools?

calcmogul, 2021-11-21:
The 2022 FRC Game Tools are under "File Releases" on TeamForge.

bovlb, 2021-11-21:
Thank you!  I followed that link and suddenly my TeamForge account has access to "FRC 2022 Beta".  I don't know why it wasn't there before.

closed 2021-11-21 by PeterJohnson