wpilibsuite/sysid #239: Make integration tests not run for vendordep configurations

calcmogul, 2021-12-04:
Few vendors have bothered to support simulation on all platforms, and others don't plan to going forward (navX). For this tool to be any use, we need support for vendors.

If we remove problematic vendors from our integration tests, we should be able to use athena-only vendordeps irl. This reduction in test coverage isn't ideal, but we've exhausted our options. At this point, it's the vendor's fault if they didn't provide us sim builds and sysid breaks for their vendordep irl.

Piphi5, 2021-12-04:
Which vendordeps should we support? I know CTRE and REV should be on the table.

calcmogul, 2021-12-04:
All of them, but we should only run integration tests for those that have sim support on all platforms.

Piphi5, 2021-12-05:
Right, is there anyway we could check that the right hardware was configure though during my the integration tests? Having an ifdef else to set the data collection functions to return zero sounds really repetitive and verbose.

calcmogul, 2021-12-05:
It's either that or we don't have integration tests at all. I don't see how else we can do it. :shrug:

PR #241 (2021-12-09): Add CTRE + ADXRS450 Support and finalize Gyro Integration Tests
Fixes #239
closed 2021-12-09 by calcmogul