wpilibsuite/sysid #240: Add ADIS Gyro Support

Piphi5, 2021-12-05:
Currently not supported in generator setup

sciencewhiz, 2021-12-05:
Has the libraries been updated for 2022? Last I checked they hadn't.

Piphi5, 2021-12-05:
I don't think so, but currently, they're not even commented out or set up as a supported gyro anywhere in the code. So this issue mainly serves as a reminder so we don't forget about it.

ngreen-gt, 2021-12-10:
I was just stopping by to ask about the ADIS16470, so glad to see it is in progress.  I did have to manually add it to our project until it is updated for 2022.

Piphi5, 2021-12-10:
What do you mean by manually adding it?

ngreen-gt, 2021-12-10:
Having to copy the source class into my project and then modifying the stuff that changed in 2022.  Here is the modified java class:  https://github.com/frc1108/2022NuggetBeta/blob/trunk/src/main/java/frc/robot/pantherlib/ADIS16470_IMU.java

Piphi5, 2021-12-10:
@calcmogul is this something we should consider if ADIS isn't updated in time?

Piphi5, 2021-12-11:
I'll consider adding it to the PR

calcmogul, 2021-12-11:
This PR is relevant: https://github.com/wpilibsuite/allwpilib/pull/3777

Piphi5, 2021-12-11:
I'll wait for that to merge

PR #242 (2022-01-03): Add ADIS IMUs
Fixes #240
closed 2022-01-03 by calcmogul