wpilibsuite/sysid #244: NEO Drivetrain Generator Code Crashing (beta3)

ngreen-gt, 2021-12-10:
I don't know if the REV stuff should be working yet, but at least it seems not to be on my end.  I've tried a couple different configs and nothing yet seems to work.  

It does deploy from the generator fine, but appears to be crashing.  I'm assuming a REVlib issue within sysid.  Since I can get the drivetrain working normally outside sysid.

I will note that it is slightly confusing with the "Data Port" vs. "Encoder Port".  It seems like the hall effect NEO sensor should be Encoder Port, but then it also give options to invert that sensor, which isn't how it should work I think.  But I have tried deploying code both ways.  I've also just went down to a single motor controller pair, though I don't think it is related to that. 

The gyro is a separate issue (ADIS) and so I've used none.  I've put the config.json in the below code.  

This could be just a matter of waiting to fix REVlib stuff in a newer release, and if that is the case, you can ignore. 

`{
  "counts per rotation": 1.0,
  "encoder type": "Data Port",
  "encoding": false,
  "gearing": 10.71,
  "gyro": "None",
  "gyro ctor": "1",
  "is drivetrain": true,
  "motor controllers": [
    "SPARK MAX (Brushless)",
    "SPARK MAX (Brushless)"
  ],
  "number of samples per average": 1,
  "primary encoder inverted": false,
  "primary encoder ports": [
    0,
    1
  ],
  "primary motor ports": [
    1,
    3
  ],
  "primary motors inverted": [
    true,
    true
  ],
  "secondary encoder inverted": false,
  "secondary encoder ports": [
    2,
    3
  ],
  "secondary motor ports": [
    2,
    4
  ],
  "secondary motors inverted": [
    false,
    false
  ],
  "velocity measurement period": 100
}`

ngreen-gt, 2021-12-10:
I should have checked pull request.  Seeing as REVlib was only 4 days ago, I assume that will correct this issue.

Piphi5, 2021-12-10:
Just to confirm, you were using beta 3 when this issue occured?

ngreen-gt, 2021-12-10:
Yes.  I tried beta 2.  Remembered about Revlib.  Then uninstalled and installed beta 3.  And then was still having it crash the code after deploying from sysid tool.

Piphi5, 2021-12-10:
Yeah that makes sense. If you want you can test this version by downloading the release executables from GitHub actions (see README)

ngreen-gt, 2021-12-10:
I did this running the latest development version `v2022.1.1-beta-3-12-gf694822`

So this time the robot code never NT connected or connected to the DS, but it also didn't fully crash so that the log was being generated.

This is the error message that kept showing up:

` /home/lvuser/frcUserProgram: error while loading shared libraries: libCTRE_Phoenix_WPI.so: cannot open shared object file: No such file or directory 
`

ngreen-gt, 2021-12-11:
I'll have to check later, but the branch I tested I think didn't have the CTRE libraries (like the 2nd one does).  So that is probably part of the cause of the errors.

Piphi5, 2021-12-11:
It's probably a CTRE bug when deploying to the RIO. If the CTRE libraries weren't included, it wouldn't cause this error.

Piphi5, 2021-12-11:
We can make sure the issue is due to the inclusion of CTRE code if you test this [build](https://github.com/wpilibsuite/sysid/actions/runs/1566329307), which is basically REV with the CTRE code removed.

ngreen-gt, 2021-12-11:
While this did deploy and I could connect & run the autonomous tests, there were a couple big issues with the build `v2022.1-1-beta-3-12-g279720f`

**Sysid stuck on test warning popup** While I could run the quasistatic & dynamic test from the drive station.  When I disabled the robot at the end of autonomous driving, the sysid warning stayed active and it didn't proceed to the next screen with encoder distances, etc.  To exit the warning you have to click `End test` and so there isn't data recorded.  The test show up as `Not run` after this.  

**Unclear or unknown encoder distance in Shuffleboard** There are a few issues here.  First, the labels in Shuffleboard seem wrong.  It is missing a `Right Velocity` however it seems that `Right Position` is displaying velocity (so it is really the position missing).  Second, the distance traveled is off by ~30x.  I went about 3m and the position reading changed by 90.  I am using the `encoder port` with a encoder CPR value of 1 (NEO motors) and a gearing of 10.71:1 (Toughbox Mini).  In the logger I have units of Meters, and a `Units per rotation` of 0.4788 (6" wheels).  The dynamic test velocity had value ~55.

calcmogul, 2021-12-11:
https://github.com/wpilibsuite/sysid/pull/246

calcmogul, 2021-12-11:
>  a gearing of 10.71:1

Is that the gearing from the motor to the wheel, or the encoder to the wheel? Those tend to be different, and SysId wants the gearing from the encoder to the wheel.

ngreen-gt, 2021-12-11:
The NEO motor encoder (the note does say use 1 for the counts per rotation) though maybe the 42 cpr is showing up for some reason.  That could make sense potentially.

From the motor to the wheel is just the AM gearbox of 10.71:1.  You probably could add to the note about the gear ratio (since I think it is slightly less clear having two fields vs. just one i.e. making me remember it is 10.71:1 and not 1:10.71)

ngreen-gt, 2021-12-11:
Looking into it, it doesn't seem like the cpr is used when using the Encoder Port, so changing that value shouldn't change the value.  I don't think it should anyhow.  https://github.com/wpilibsuite/sysid/blob/bdafa3146247d909707dc9bd9605f1c612c3f032/sysid-library/src/main/cpp/generation/SysIdSetup.cpp#L236

Another little issue is that the Spark brushless encoder port shouldn't have the option to invert it in the generator.  I think that should go with the the data port (Alternate encoder).

Piphi5, 2021-12-11:
That could be the issue

Piphi5, 2021-12-12:
[This](https://github.com/wpilibsuite/sysid/actions/runs/1568513211) should hopefully fix that issue atleast.

ngreen-gt, 2021-12-12:
I'm not exactly where this went wrong but it makes less sense to me somehow now.  

1) Should the position value be anything after the robot generator code is first deployed and when the robot code is restarted?  It has been some number.  It may not matter to much, but maybe it is something to reset the encoder value somewhere.  Right now it seems like a junk number (see picture).

2) The numbers got a lot larger in this newest revision.  Now the position values are millions of units.  I did try to alter the "cpr" but I couldn't make sense but didn't seem to change the position output.  It is too challenging to see the velocity because the number extended even further (maybe in billions.  

[image]

Piphi5, 2021-12-12:
[This](https://github.com/wpilibsuite/sysid/actions/runs/1570497442) should fix that issue

ngreen-gt, 2021-12-12:
On start the position value was -0.95.   A lot closer to zero, but not quite.  Moving around it was still slightly too much.  Like the velocity was ~5 when doing the dynamic test, when this drivetrain max velocity was ~3 m/s.

Piphi5, 2021-12-12:
Are you referring to the smart dashboard values?

ngreen-gt, 2021-12-12:
Yes.  That is where I'm looking at values.

BTW none of the versions are recording the values when the robot disables.  When disabling after having the robot auto drive during the test, then the sysid program is just stuck.  You can hit the end button to revive the program, but that seems to discard whatever data might have been recorded.

ngreen-gt, 2021-12-12:
I have to go back and see what version was still working I think because it did work in earlier versions when doing the talon tach stuff.

Piphi5, 2021-12-12:
Smart dashboard has rotations and so if you said you had around 5 rotations/s that should roughly translate to 2.5 m/s so I think that issue is resolved. Now we just have to look into why the telemetry things aren't working.

Piphi5, 2021-12-12:
Was the talon and tach stuff during 2021?

ngreen-gt, 2021-12-12:
Had to be.  It wasn't the new RoboRIO firmware.

Piphi5, 2021-12-13:
Does the GUI say anything along the lines of: `TelemetryManager did not receieve data 5 seconds after `

ngreen-gt, 2021-12-13:
No it doesn't. I think the NT switching to disabled makes some sense, but maybe not totally clear. I guess check all the conditions in the if to be able to switch to disabled too.

Piphi5, 2021-12-13:
The disabling stuff should have gotten fixed on wpilib's end so [this build](https://github.com/wpilibsuite/sysid/actions/runs/1574875403) should work.

Piphi5, 2021-12-13:
Actually, use [this build](https://github.com/wpilibsuite/sysid/actions/runs/1574978245) instead

ngreen-gt, 2021-12-17:
I tried [this build](https://github.com/wpilibsuite/sysid/actions/runs/1585756372) and the FMS issue was resolved in it.

closed 2021-12-19 by ngreen-gt