wpilibsuite/sysid #258: Support FeedbackControllerPresets based on connected Encoder

modelmat, 2021-12-19:
https://github.com/wpilibsuite/sysid/pull/253#discussion_r771232288

Currently REV's brushless controller internal encoder uses a hall effect sensor with custom, unchangeable settings:

[image]

However, using the brushless mode on the motor controller apparently does not require to use the internal encoder, so using that setting may sometimes be wrong for the gains, hence it will be necessary to change the FeedbackControllerPreset based on the connected encoder.

Piphi5, 2021-12-31:
Waiting on REV engineers to confirm the right measurement delay for the NEO encoders (sometime next week) before fixing this issue

Piphi5, 2022-01-12:
For the sake of documenting this, here's the official response from REV Support (Note: REV is working on lowering this latency so this may be subject to change in the future):

```
 That sensor is sampled every 32ms, and the sampling window is 8 samples, which means that the sampling window is 256ms. Also, the hall sensor decoding does not use a backward finite difference (unlike the quadrature encoder decoding), so I believe the formula that should be used here is (N - 1)/2 * T.

(8 - 1)/2 * 32 = 112ms

```

PR #306 (2022-01-14): Create proper SparkMax Presets
Fixes #258
closed 2022-01-14 by calcmogul