wpilibsuite/sysid #269: Refactor plot display code

Piphi5, 2021-12-25:
Currently, there is a lot of [hard coding](https://github.com/wpilibsuite/sysid/pull/268#pullrequestreview-839996036) that makes it difficult to customize the order of the plots if need be.

Also, the code could probably be cleaned up alot

ngreen-gt, 2021-12-26:
I wanted to piggyback here that it would be nice if plots were also resettable.  The workspace>reset doesn't seem to affect the plots.

Piphi5, 2021-12-26:
So like a button to reset the plot fittings?

ngreen-gt, 2021-12-26:
How about if the analyzer reset button would also clear the diagnostic plots?  I'm not how compartmentalized the windows are but that would seem to make sense to me.  It is useful only that if you were to analyze a new file that you'd know that new diagnostic plots were generated for sure.

calcmogul, 2021-12-26:
https://github.com/wpilibsuite/sysid/pull/270

PR #339 (2022-01-30): Refactor plots
The voltage domain plots were removed since they aren't relevant to the
fit anymore. The acceleration plots were removed since they are almost
always noise and aren't helpful for troubleshooting.

The GUI layout was reorganized so all the plots fit in the main display
and the only windows with a scrollbar are the generator and program log.

Instances of "units" are now replaced with an abbreviated unit name so
they use less horizontal space.

Fixes #269.
closed 2022-01-30 by calcmogul