wpilibsuite/sysid #278: SysID doesn't work if used on a freshly imaged roboRIO

jasondaming, 2021-12-28:
If you image a roboRIO and try to use SysID it will "complete" the deploy but there will be no robot code running and you can't connect to it.  I couldn't even see any logs to try to figure out what the issue was.

The easy fix to this is to deploy ANY frc program to the RIO and then use SysID again and it will work fine.

calcmogul, 2021-12-28:
It's probably not deploying all the libraries it needs to. We should compare what libraries are deployed via GradleRIO vs SysId.

ThadHouse, 2021-12-28:
I explicitly made it not include opencv. But that shouldn't affect things. I'll check when I get home.

ThadHouse, 2021-12-31:
Ah, this is the "exec" bug

ThadHouse, 2021-12-31:
https://github.com/wpilibsuite/GradleRIO/blob/main/src/main/java/edu/wpi/first/gradlerio/deploy/roborio/RoboRIO.java#L49

PR #291 (2022-01-04): Workaround exec bug on deploy
Fixes #278.
closed 2022-01-04 by calcmogul