wpilibsuite/sysid #288: Reuse the gearing and analysis type in the Logger

jasondaming, 2021-12-31:
Analysis Type and gearing could be reused in the Logger application to reduce some of the math and make user's lives easier.

The units per rotation would have to be changed to something like (hopefully someone can come up with something shorter) "Post Gearing units per rotation".

jasondaming, 2021-12-31:
If this change is not made / accepted we should change the tooltip on Units per rotation.

randomstring, 2021-12-31:
"Post Gearing units per rotation" is still ambiguous. Is this per rotation of the encoder or motor? In most cases these are one and the same, but not always. 

A concrete example would help. 6" diameter wheel,  8.16:1 gear ratio between encoder:  (6" * pi / 8.16) / 39.3701 (inches/m) = .058 meters

jasondaming, 2021-12-31:
That is why we have the documentation which does have an example.

Piphi5, 2022-01-04:
Ideally, the robot is reporting output shaft rotations, meaning that the gearing would be accounted for in the project configuration step. The units per rotation in the logger is simply to convert that output shaft rotation to whatever units the user wants to use for the analysis. I think it's the documentation on the frc-docs website that may be off as it also includes the gearing calculation in the units per rotation for the logger.

As for the analysis type, generation is mainly looking at 3 types relevant for which project to use (Romi, Drivetrain, and General Mechanism). The logger uses more specific test types (Elevator, Arm, etc.)

 So I don't really think the original issue stands, but it definitely seems like a lot of things aren't clear about the logger and the generator. What changes should we make to the layout/tooltips/docs to sort of clarify the previously mentioned workflow?

Piphi5, 2022-01-09:
So the main thing to close this issue out is to clarify the ambiguity of the units per rotation tooltip. Make it clear that its units of every output shaft rotation rather than reincluding the gearing in that calculation.

PR #304 (2022-01-14): Clarify units per rotation tooltip
Fixes #288
closed 2022-01-14 by calcmogul