wpilibsuite/sysid #296: Implement stairstep data collection test

calcmogul, 2022-01-04:
This would step the voltage by a predetermined amount, step it again when the velocity reaches steady-state, then repeat. A max distance could be implemented so the test knows when to start going backward. To determine steady-state, we could have a running variance of the acceleration based on code like this:
```cpp
double sysid::GetAccelNoiseFloor(const std::vector<PreparedData>& data,
                                 int window) {
  double sum = 0.0;
  size_t step = window / 2;
  auto averageFilter = frc::LinearFilter<double>::MovingAverage(window);
  for (size_t i = 0; i < data.size(); i++) {
    double mean = averageFilter.Calculate(data[i].acceleration);
    if (i >= step) {
      sum += std::pow(data[i - step].acceleration - mean, 2);
    }
  }
  return std::sqrt(sum / (data.size() - step));
}
```
where steady-state occurs when the acceleration drops below that noise floor.

This one test would replace all four of the existing tests since it sufficiently samples across the whole velocity-acceleration plane. Ks, Kv, and Ka are uniquely determined because Ka is determined by the transient, Ks manifests as a constant bias in the steady-state, and Kv manifests as a linearly increasing bias in the steady-state.

Piphi5, 2022-01-04:
I think they should be able to choose between a maximum distance or maximum duration for a test like a flywheel where distance is less trivial.

Oblarg, 2022-01-11:
another possible feature: selectable option to run two tests (both directions) or one.

calcmogul, 2022-01-11:
When would you use one or the other? I don't like giving users more choices unless there's a valid reason.

Piphi5, 2022-01-11:
We probably want backwards data.

We could have a forwards and backwards test so that the user can stop it manually instead of automatically?

calcmogul, 2022-01-11:
Yea, two manual tests is a bit safer.

Piphi5, 2022-01-15:
Current [procedure](https://github.com/Piphi5/sysid/tree/step-test):

If abs(acceleration) <= 0.2 -> wait one second -> step voltage -> if abs(acceleration) >= 0.2 -> repeat

Will try to implement the logging and analysis side of things in the next few days

calcmogul, 2022-01-19:
This procedure should allow getting rid of the motion threshold entirely because the robot will immediately start moving.

Piphi5, 2022-01-19:
It may not since the initial step voltage might not be enough to cause motion.

calcmogul, 2022-01-19:
Idk how much the quasistatic data trimming affects the fit.