wpilibsuite/sysid #312: ImGUI crashes on JSON with data below motion threshold

calcmogul, 2022-01-17:
Here's the JSON.
[sysid_data20220117-182153.json.txt](https://github.com/wpilibsuite/sysid/files/7883734/sysid_data20220117-182153.json.txt)

The first run opens a dialog saying the quasistatic test had all the data trimmed. Here's the logs.
```
> Task :sysid-application:run
INFO: Opening File: /tmp/sysid_data20220117-182153.json (Analyzer.cpp:490)
INFO: Opened File (Analyzer.cpp:494)
INFO: Read /tmp/sysid_data20220117-182153.json (AnalysisManager.cpp:573)
INFO: Parsing initial data of /tmp/sysid_data20220117-182153.json (AnalysisManager.cpp:594)
INFO: Preparing Data. (Analyzer.cpp:510)
INFO: Preparing Drivetrain data (AnalysisManager.cpp:608)
INFO: Reading JSON data. (AnalysisManager.cpp:431)
INFO: Preprocessing raw data. (AnalysisManager.cpp:438)
INFO: Copying raw data. (AnalysisManager.cpp:450)
INFO: Converting to PreparedData struct. (AnalysisManager.cpp:454)
INFO: Applying trimming and filtering. (AnalysisManager.cpp:501)
INFO: Calculating Gains. (Analyzer.cpp:525)
INFO: Returning early for gain calculation. (Analyzer.cpp:527)
INFO: Graphing Data. (Analyzer.cpp:552)
INFO: Attempting to graph only raw time series. (Analyzer.cpp:556)
```
When SysId is closed and reopened, it crashes with the following exception from ImGUI.
```
> Task :sysid-application:run
INFO: Previous file location exists: /tmp/sysid_data20220117-182153.json (Analyzer.cpp:81)
INFO: Read /tmp/sysid_data20220117-182153.json (AnalysisManager.cpp:573)
INFO: Parsing initial data of /tmp/sysid_data20220117-182153.json (AnalysisManager.cpp:594)
INFO: Preparing Data. (Analyzer.cpp:510)
INFO: Preparing Drivetrain data (AnalysisManager.cpp:608)
INFO: Reading JSON data. (AnalysisManager.cpp:431)
INFO: Preprocessing raw data. (AnalysisManager.cpp:438)
INFO: Copying raw data. (AnalysisManager.cpp:450)
INFO: Converting to PreparedData struct. (AnalysisManager.cpp:454)
INFO: Applying trimming and filtering. (AnalysisManager.cpp:501)
INFO: Calculating Gains. (Analyzer.cpp:525)
INFO: Returning early for gain calculation. (Analyzer.cpp:527)
INFO: Preparing Data. (Analyzer.cpp:510)
INFO: Returning early for data preparation. (Analyzer.cpp:512)
INFO: Graphing Data. (Analyzer.cpp:552)
INFO: Attempting to graph only raw time series. (Analyzer.cpp:556)
ERROR: An error occurred when attempting to load the previous JSON. (Analyzer.cpp:105)
sysid: /__w/thirdparty-imgui/thirdparty-imgui/imgui/imgui.cpp:6298: bool ImGui::Begin(const char*, bool*, ImGuiWindowFlags): Assertion `blocking_modal != __null' failed.
```

There's two issues here. The first is that the motion threshold is hardcoded to "units/sec", which wasn't appropriate for this data. The second is even if that setting is incorrect, the GUI shouldn't crash.

Piphi5, 2022-01-18:
Are you saying we should calculate some noise floor and automatically determine the default motion threshold?

calcmogul, 2022-01-18:
I don't know what the best fix is, but what we have now is broken in some scenarios.

danielml2, 2022-01-18:
Hey, I'm the original guy this issue has happened to.
A friend of mine just sent this to someone he knew contributing to SysId

since then i've tried changing both the quasistic ramp rate to 0.2 and the units per rotation to 0.2 trying to see if it would change anything, but again same result happens with the "Quasistic test trimmed data" error message
the GUI crashing if i try to change anything in the analyzer, and all the results being nulled to -9999

Here's the JSON file from when i changed the settings to 0.2
[sysid_data20220118-191843.json.txt](https://github.com/wpilibsuite/sysid/files/7891546/sysid_data20220118-191843.json.txt)

Program Log shows the same when using the first JSON file.

Seems to me the tool is unusable for me, I'm using the latest version (2022.2.1) on Windows 10.
roboRIO's image has been updated to 2022's image aswell.

Would love to know if this issue has been happening to anyone else, or if its an issue on my end.

calcmogul, 2022-01-18:
That data also has really small values. SysId is able to generate plots for that data with a motion threshold of 0.001. Do you have a generator config? Something is very wrong with your units.

danielml2, 2022-01-18:
Here's the generator config file
[config.json.txt](https://github.com/wpilibsuite/sysid/files/7891992/config.json.txt)

In the original robot code, we have only one of the right motors inverted, didn't know if I should've ticked the box to invert all of the right side encoders, or not either way i've tested both before changing the values to 0.2

Each side of the robot has a TalonSRX and a VictorSPX motor
Pigeon IMU for the Gyro
and used originally around ~0.63 for the units per rotation value.
Gearing is 8 to 1, so I've set it up as 8:1 in the GUI (wasn't sure on which one was right so I've tried doing the opposite too, didn't change by a lot)

calcmogul, 2022-01-19:
SysId's deployed program lets you drive the robot around in teleop. Watch NetworkTables to make sure the values being reported make sense for the units you chose.

Piphi5, 2022-01-19:
You can also just deploy a SysId project and push the robot

calcmogul, 2022-01-19:
The referenced PR fixes the crash, but we still need to figure out why the data values in question are so small.

closed 2022-02-02 by calcmogul