wpilibsuite/sysid #317: Assertion Failed: IMGUI Error and

Ocean-Moist, 2022-01-22:
Every time I try to open SysID I get this error:
[image]
When using the main branch and building from source I get a different error, this time after running any tests (i.e quasistatic forward) and pressing disable on the driver station.  It says cannot seek string iterator after end.

PeterJohnson, 2022-01-22:
Delete C:\Users\<username>\AppData\Roaming\sysid*.json

Ocean-Moist, 2022-01-22:
Ok, I can open it, but I still get the string iterator error. 

[image]

PeterJohnson, 2022-01-22:
Is this on trying to analyze a particular file?  Posting the file would enable us to reproduce the issue to track down the root cause.

Ocean-Moist, 2022-01-22:
While running the Quasistatic tests, this error occurs after I disable from the driver station.
here is my generator config:
https://pastebin.com/5kHqehc4
here is my logger data:
https://pastebin.com/Be1gfF0f

Ocean-Moist, 2022-01-22:
Now after leaving it open I am getting another error. I just idled in Sys ID and this happened:
[image]

PeterJohnson, 2022-01-23:
I have no idea how it's tripping that assertion.

nolanhergert, 2022-02-03:
> For this error, delete C:\Users<username>\AppData\Roaming\sysid*.json. The error you're seeing with tests is different. Does that error give you a source file / line number?

In my case, the sysid*.json files are defaulting to a different folder (C:\Users\Public\wpilib\2022\vscode). When I move the json files into a different folder, then I can launch the system identification tool again. Until I run into #319 and close normally. Then it shows up again!

Do you know why it's aborting? I would like to avoid having to do this each time / explain it to the students. The files are here: https://www.dropbox.com/s/ydn8b22x1r01odb/subfolder.zip?dl=0

calcmogul, 2022-02-03:
The recent release (2022.3.1) fixes a lot of assertion failures.

nolanhergert, 2022-02-03:
I no longer see this error! Now, just to get #319 working, but I suspect (due to the updated error message) that it's a problem with my setup. Thanks!

closed 2022-02-03 by calcmogul