wpilibsuite/sysid #318: Incorrect Position measurement using CANCoder

Sheraz4201, 2022-01-23:
We were testing Sysid on our turret, which is a VictorSPX controlling a 775 motor with a CANCoder as its remote sensor. In our results, we saw that the position value returned to SmartDashboard was way off even when not running any of the 4 tests. The position was reporting a number like 4e+60 as our turret's current position. Checking the values the CANCoder returned using Phoenix Tuner showed that it was reporting a correct value when we manually turned our turret.  We are using the latest version of Phoenix Tuner with the latest CANCoder firmware (22.0.0.0). I have attached the configuration we used and the data that our team collected from running the characterization tool on our turret.

[turretconfig - Copy.txt](https://github.com/wpilibsuite/sysid/files/7919520/turretconfig.-.Copy.txt)
[sysid_data20220122-143225 - Copy.txt](https://github.com/wpilibsuite/sysid/files/7919521/sysid_data20220122-143225.-.Copy.txt)

Piphi5, 2022-01-23:
We forgot to pass in the gearing by value so it's becoming a garbage value, I'll fix this tonight.

Piphi5, 2022-01-23:
@Sheraz4201 can you see if the executables [here](https://github.com/wpilibsuite/sysid/actions/runs/1734820310) fix this issue?

Sheraz4201, 2022-01-23:
Thanks, I'll test it tomorrow.

Sheraz4201, 2022-01-23:
The new changes made it so that when we physically moved the turret 90 degrees, it recorded a change of 1 position unit in Smartdashboard. Is that correct?

Piphi5, 2022-01-23:
Don't think so. The units in smart dashboard / shuffleboard, etc are in rotations. 90 degrees should respond to .25 position units. Judging by the config file however, your gearing may be off. It's a gearing of 18:120, meaning you are gearing up a turret and with a motor like a 775 I'm not sure it would have enough torque to push the motor, and the motor would be very difficult to control. I'm assuming you meant 120:18?

Sheraz4201, 2022-01-23:
Oh yeah I did, I'll test it soon.

calcmogul, 2022-01-26:
Reopening because that PR fixed half of the issues raised by this PR so it needed to be merged, but unlinking PRs isn't possible.

Piphi5, 2022-01-26:
@Sheraz4201  any updates on this?

Sheraz4201, 2022-01-26:
Sorry, I got caught up in some other stuff and I forgot about this. I'll be able to test it on friday.

Piphi5, 2022-01-27:
No worries

Sheraz4201, 2022-01-29:
We think we had our gear ratio wrong, and we tried it with a different gear ratio, and it seemed to work okay but we're still testing it. Our gear setup is also bit different from other teams. We have 2 different gear ratios, since there's a gearbox (27:1) between the motor and CANcoder and a chain between the CANcoder and the actual turret (120:18). I'll be testing more tomorrow so I'll let you know how it goes.

Piphi5, 2022-02-03:
Any updates on this?

Sheraz4201, 2022-02-03:
Apologies for not responding. We were able to successfully characterize the turret using a gearing/gear ratio of 1:4.05 (4.05 is the gear ratio of our entire turret (27:1 * 18:120)). We also changed the counts per revolution of the CANcoder from 4096 to 360 since we found that the CANcoder returns values in degrees by default if you read its position directly, which is something that we did not account for before.

Piphi5, 2022-02-03:
No worries, so can this issue be closed?

Sheraz4201, 2022-02-04:
Oh uhh yes it can.

PR #320 (2022-01-26): Pass CANCoder Gearing by value
Fixes #318
closed 2022-02-04 by calcmogul