wpilibsuite/sysid #319: Error: Quasistatic test trimming removed all data

Sentientnapkin, 2022-01-23:
Here is the file I am trying to analyze: https://cdn.discordapp.com/attachments/900608549552529429/934642152472977428/sysid_data20220122-085328.json

Piphi5, 2022-01-23:
When this error occurs, you can lower the Velocity threshold as [mentioned here](https://docs.wpilib.org/en/stable/docs/software/pathplanning/robot-characterization/viewing-diagnostics.html#improperly-set-motion-threshold).

The graphs load with a lowered threshold, however, judging by the fact that its a drivetrain test, and the units are technically impossible (1 rotation != .4710 rotations), I assume there is a major issue with how you are setting your units and units per rotation:
[image]

[image]

Piphi5, 2022-01-23:
@Sentientnapkin can you send the generation config you used?

Sentientnapkin, 2022-01-23:
So, I fixed the config (https://pastebin.com/nVSwmLnq), and it still tells me the same error. Here is the data we collected: https://gist.github.com/Sentientnapkin/82617e615f9ed8e846691af486f9c9a5

Piphi5, 2022-01-23:
I think the gearing is wrong. It says that you have a 1:6 gear ratio, which you probably meant to have as a 6:1 since that's gearing the drivetrain down.

Sentientnapkin, 2022-01-23:
> I think the gearing is wrong. It says that you have a 1:6 gear ratio, which you probably meant to have as a 6:1 since that's gearing the drivetrain down.

for every six turns of the motor, the wheel turns once. but we are using REV throubore encoders that are directly on the out put shaft. so when the wheel turns once, the encoder would have 8192 counts, or one revolution, while the motor would have spun 6 times. what are the appropriate settings in this scenario

Piphi5, 2022-01-23:
The gearing is between the encoder shaft and output shaft. If the encoder is on the output shaft, the gearing is 1:1

Piphi5, 2022-01-23:
Also judging by these plots of your data, you don't seem to move during the slow-forward or fast-forward tests:
[image]
[image]

It seems like the robot doesn't send forward commands to the motors:
[image]

Sentientnapkin, 2022-01-23:
well the robot does actually move, does this mean I have encoder ports wrong?

Sentientnapkin, 2022-01-23:
I do not understand why it worked for the other tests if I got encoder ports wrong. During slow forward and Dynamic forward it does move.

Piphi5, 2022-01-23:
Its most likely not a port issue or the program would've crashed.

Piphi5, 2022-01-23:
It seems like the tests are very short-lived. How long do you leave the slow forward and dynamic forward running for?

Sentientnapkin, 2022-01-23:
around 30 feet.

Piphi5, 2022-01-23:
What does sysid report measuring in the telemetry window?

Sentientnapkin, 2022-01-23:
received data with size 3587 for the slow-forward test in 17 seconds.

Piphi5, 2022-01-23:
This initial popup also changes after the test to say how far the mechanism moved:
[image]

What distance does it read after you hit disabled?

Piphi5, 2022-01-23:
> received data with size 3587 for the slow-forward test in 17 seconds.

Is this with respect to the bad data you sent or is this a new test?

Sentientnapkin, 2022-01-23:
Like 1.98

> > received data with size 3587 for the slow-forward test in 17 seconds.
> 
> Is this with respect to the bad data you sent or is this a new test?

bad data. and it said 1.98.

Piphi5, 2022-01-23:
So SysId noticed that the robot moved 1.98 meters but then saved bad data?

Sentientnapkin, 2022-01-23:
I redid it and got no error. But my data is off by a factor of like 3. here is my data: https://gist.github.com/Sentientnapkin/1358fdd3799d900455c4bf1a2d78fc1d. It definitely goes faster than .7 m/s during the dynamic tests.

Piphi5, 2022-01-24:
What config are you using now?

Sentientnapkin, 2022-01-24:
{
  "counts per rotation": 256.0,
  "encoder type": "roboRIO quadrature",
  "encoding": true,
  "gearing denominator": 28.0,
  "gearing numerator": 3.0,
  "gyro": "NavX",
  "gyro ctor": "SerialPort.kMXP",
  "is drivetrain": true,
  "motor controllers": [
    "VictorSPX",
    "VictorSPX"
  ],
  "number of samples per average": 5,
  "primary encoder inverted": true,
  "primary encoder ports": [
    0,
    1
  ],
  "primary motor ports": [
    0,
    1
  ],
  "primary motors inverted": [
    false,
    false
  ],
  "secondary encoder inverted": false,
  "secondary encoder ports": [
    2,
    3
  ],
  "secondary motor ports": [
    2,
    3
  ],
  "secondary motors inverted": [
    true,
    true
  ],
  "velocity measurement period": 100
}

Piphi5, 2022-01-24:
Why are your counts per rotation 256 now? And again the gearing is a little concerning. The gearing numerator should be bigger than the denominator.

Sentientnapkin, 2022-01-24:
The gearing is incorrect, I accidentally uploaded the wrong file, it's 28:3, everything else is right. We are using a different robot to test if it works and the CPR is 256 on this one.

Piphi5, 2022-01-24:
so was the gearing correct when you tested the robot?

closed 2022-02-02 by calcmogul