wpilibsuite/sysid #324: Sysid breaking when using drivetrain (angular) parameter

ctrl-alt-delete101, 2022-01-25:
I tried characterizing our team's practice robot to get angular gains from SysID. I was able to run all the tests properly however when trying to read the data I got an exception error message: "Quasistatic test trimming removed all data". 
[image]

After getting that error message I exited sysid and tried to reload it but I was given another error
[image]

Json file that I used
https://gist.github.com/ctrl-alt-delete101/edcb94982856901b68584a63212e3aab

Piphi5, 2022-01-25:
[image]
[image]

From this script, it seems like the robot didn't move a whole lot. Was that the case?

ctrl-alt-delete101, 2022-01-25:
Yea the robot didn't rotate for that long, however I did the same with a longer trial and the same thing occured.

ctrl-alt-delete101, 2022-01-25:
> [image] [image]
> 
> From this script, it seems like the robot didn't move a whole lot. Was that the case?

Yea the robot didn't rotate for that long, however I did the same with a longer trial and the same thing occured.

Piphi5, 2022-01-25:
Can you send data for the longer trial?

ctrl-alt-delete101, 2022-01-25:
I don't have that data on me right now. Although I don't know if its useful but I have data from a normal drivetrain parameter.

Piphi5, 2022-01-25:
Do you know what gearing you used?

ctrl-alt-delete101, 2022-01-25:
I believe its 16:1

Piphi5, 2022-01-25:
Did the reported measured values make sense to you?

ctrl-alt-delete101, 2022-01-25:
Everything seemed relatively normal, however the acceleration graphs looked weird. I tried to use the values on our mecanum drive but the wheels kept going back and forth so I believe I have to correctly tune the position controllers. I'll also link the json file that I used.

r-squared: 0.99995
Sim rsme: 0.12369

https://gist.github.com/ctrl-alt-delete101/146fe418d9ad6a410a725ad5d90d0a05

Piphi5, 2022-01-26:
FYI, the sysid on main is getting a better fit. Can we consider this issue closed?
[image]

ctrl-alt-delete101, 2022-01-27:
Yep, seems good.

closed 2022-01-27 by ctrl-alt-delete101