wpilibsuite/sysid #326: Editing Projects with Custom Code

LudumDareDevelopment, 2022-01-25:
Is it possible to edit the code that is deployed to the robot from sysID like frc-characterization could? I plan on characterizing a swerve drive, and it would be nice if I could command my steering motors to an angle of 0 constantly, but I currently cannot set this up with sysID.

Daltz333, 2022-01-25:
Unfortunately not. This is a design decision. Editing the custom project has led to more issues than benefits.

calcmogul, 2022-01-25:
It comes down to not being able to trust the average user to not mess things up.

LudumDareDevelopment, 2022-01-25:
Interesting. Thank you for the help!

LudumDareDevelopment, 2022-01-25:
Is frc-characterization forwards compatible with WPILib 2022? I would imagine it would still work if the Network Tables protocol did not change

Piphi5, 2022-01-25:
No the network tables protocol is different

Piphi5, 2022-01-25:
If you know cpp you can use the files in sysid-library to take advantage of the new protocol

calcmogul, 2022-01-25:
However, if you record data with frc-characterization, it'll be transparently converted to the sysid format if it's loaded into sysid.

LudumDareDevelopment, 2022-01-25:
Would recording data with frc-characterization be possible if the characterization project version is set to WPILib 2022?

calcmogul, 2022-01-25:
I don't believe so.

closed 2022-01-25 by LudumDareDevelopment