wpilibsuite/sysid #331: Pigeon Yaw units are double-converted from radians to degrees

jlmcmchl, 2022-01-26:
Tested w/ version 2022.1.1 - checking commits between that release and master I've observed that the AngularDriveSim analysis has been removed, but no changes have been made to the way data is acquired from the pigeon. Due to #324 I'm unable to determine if this is an issue strictly in displaying the value or something else. 

Nevertheless - the CTRE pigeon returns a value in degrees, and that is displayed back to the user as if it had been radians, and converted once more to degrees.
[image]

As you can see, the robot turned 21804 degrees, which is 180 / pi * 380. In reality, the robot performed a little over 1 full rotation.

Piphi5, 2022-01-26:
That is definitely an issue that hasn't been fixed yet.

Piphi5, 2022-01-26:
Just confirmed pigeon gyro positional units are returned as degrees.

Piphi5, 2022-01-26:
@jlmcmchl Can you see if the executables from this [PR fixes ](https://github.com/wpilibsuite/sysid/actions/runs/1753081476) the issue?

jlmcmchl, 2022-01-26:
I'm back in the shop friday, so I'll update you after testing at that time.

Thanks for getting back so quickly!

PR #338 (2022-01-27): Fix pigeon position reporting and clean up gyro lambdas
Fixes #331
closed 2022-01-27 by calcmogul