wpilibsuite/sysid #350: Strange Analyzer graphs for Backward dataset

chauser, 2022-02-01:
The changes made an hour ago in PR #349 eliminate the chance of ever seeing this again, but I wanted to let the devs know that I am seeing some very strange graphs when looking at only the Backward dataset of a sysid drivetrain data file. The situation is complicated because this file was generated by our own deployed robot code and not by a deployed sysid logger program. And further complicated because these data are taken from running the code in simulation and not on a real robot. So it could be our fault. However, I can find nothing obviously wrong in the backward data sections of the file, and I just want to make sure that this is not indicating a filtering problem that would affect the Analysis results for Combined data.
All that said, here are three combined-diagnostics images for the Combined, Forward, and Backward datasets.
Combined
[image]
Forward
[image]
Backward
[image]

Specific things that seem odd in the Backward graphs going from left to right, top to bottom:

1. First graph: data points are shown with positive voltage and velocity. The Fit data line is horizontally displaced from the data.
2. Second graph: The acceleration portion voltage and the acceleration are positive. I would expect negative values on the voltage scale here and negative accelerations. Also the magnitude of the acceleration for corresponding to a given voltage doens't make sense when compared with the other graphs.
3 and 5. The simulation data fits the filtered data much worse than on the Forward and Combined graphs

The data file (renamed .txt because of filename upload restriction by github).
[sysid_data20220131-120114.txt](https://github.com/wpilibsuite/sysid/files/7982483/sysid_data20220131-120114.txt)

As I mentioned above, if this is merely a display problem it is mooted by the changes made today. If it is due to our mistake in generating the data file I apologize for taking your time. Normally before generating an issue, I would want to try it with data generated by sysid's own logging code but I don't have access to any such data or any way to generate some at the moment.

Thanks.

chauser, 2022-02-02:
Looking again at the graphs a few hours later I noticed that the time axes on the bottom three graphs of the Backward dataset make no sense: the backward dynamic data were gathered (from the Combined dataset graphs) with timestamps from ~153 - ~156 seconds; the axes on first two of the Backward dataset graphs show timestamps from 70-82 seconds which corresponds to the Forward Quasistatic collection period. And the last shows 70-116 seconds which corresponds to the Forward and Back Quasistatic collection periods. Maybe this will suggest something about what could be wrong to someone more familiar with this code than I am.

calcmogul, 2022-02-03:
#353's backwards graphs look fine.
[image]

closed 2022-02-03 by calcmogul