wpilibsuite/sysid #361: Add back ability to generate left/right combined drivetrain feedforward values

camearle20, 2022-02-07:
#341 removed the combo box which selected the type of analysis to do on a drivetrain characterization dataset.  While I agree that it makes sense to remove the forward/backward confusion, I still think there is value in being able to obtain a separate left and right dataset.  I think the default value of the box could remain as "combined" but there is real value in being able to produce separate left and right (combined) values.

calcmogul, 2022-02-07:
> I still think there is value in being able to obtain a separate left and right dataset.

Why?

camearle20, 2022-02-07:
Mainly for a drivetrain with less than ideal construction.  I have found it very helpful to get at least separate Ks values for each side of the drive to account for those differences.  We typically run a separate model for the left and right sides of the drive.

FWIW, I'm also seeing slightly better R2 velocity values using left combined/right combined on the previous release than using the whole combined mode.

calcmogul, 2022-02-07:
That PR removed forward vs back split. SysId never supported left vs right.

camearle20, 2022-02-07:
[image]

The options were at least in the dropdown, and clicking them gave different outputs.  I'm not sure how/if they were actually implemented though

PR #362 (2022-02-09): Display right and left datasets
Fixes #361
closed 2022-02-09 by calcmogul