wpilibsuite/sysid #363: Improve Telemetry Manager to avoid data mismatch issues

Piphi5, 2022-02-10:
Implementing features like Sequence numbers and better data polling can fix the issue where users sometimes have mismatched test data from sysid grabbing outdated Telemetry data.

Oblarg, 2022-02-12:
A flat data structure with a "test name" field makes the most sense to me.

Piphi5, 2022-02-12:
Yeah that seems like a more reasonable structure of doing things

Piphi5, 2022-02-14:
A major concern is that there right now there isn't a great way of reproducing the issue consistently. And we can't figure out if this issue is solved without that

Piphi5, 2022-02-16:
Due to potential conflicts between server and client-side handling of data we should use acknowledgment numbers to make sure data has been received and stored before clearing it.

I think the best and most testable/robust approach would be a combination of including the test on the robot telemetry output and implementing the acknowledgment numbers, resolving both test mismatches and NT issues.

PR #389 (2022-03-07): Use telemetry acknowledgement numbers, test names
Fixes #363 
For more robust data handling of the telemetry manager, the tests that
data is being collected for are included in the telemetry string.

To ensure proper communication between logger and robot, acknowledgement
numbers are used to make sure data has been properly received before
modifying it.
closed 2022-03-07 by calcmogul