wpilibsuite/sysid #364: Can't save generator configuration

itsmichaelyu, 2022-02-11:
I'm on the latest version of WPILib and have v2022.3.1 of SysID. I am also on MacOS and have both JDK 11 and Xcode Command Line tools installed. After I configure the motors, encoder, and gyro setting in the generator the save button does nothing for me; there is no prompt for me to save the file, and as far as I can tell the data isn't saved in any other preference file.

calcmogul, 2022-02-11:
What version of macOS?

itsmichaelyu, 2022-02-11:
I'm on the latest version of Monterey, which I think is 12.2.1. I also am on an intel Mac if that helps at all.

calcmogul, 2022-02-11:
We haven't tested anything on macOS 12+, so I'm not surprised it's broken. None of SysId's active developers have a mac.

itsmichaelyu, 2022-02-11:
Ah, is there any logs or anything I could do to help?

calcmogul, 2022-02-11:
This kind of thing doesn't sound like it would show up in any logs. On Linux, the dialogs won't open if you don't have a package like zenity or kdialog installed. I have no idea if macOS has a similar requirement, or if the API for that changed in macOS 12.

itsmichaelyu, 2022-02-13:
I tried the latest build and it didn't work. Interestingly enough on the latest version with WPILib all the saving functions work, except for the generator, so it's only the generator saving code that doesn't work on Mac.

Piphi5, 2022-02-15:
All the saving functions as in the telemetry / logger save button?

itsmichaelyu, 2022-02-15:
Logger and workspace saving both work, while the generator saving function doesn't work. I would think that the code for one of these could just be used to replace the generator saving code, but I am not too familiar with how exactly the program works.

calcmogul, 2023-09-21:
OBE by #518.

closed 2023-09-21 by calcmogul