wpilibsuite/sysid #383: Code not deploying

neatteamsoftware, 2022-02-22:
It says "Deploy Complete", but Driver station doesn't show that the code was received.
Here's the deploy status log:
[image]
[image]
Here's our config.json:
```
{
  "counts per rotation": 200.0,
  "encoder type": "roboRIO quadrature",
  "encoding": false,
  "gearing denominator": 1.0,
  "gearing numerator": 1.0,
  "gyro": "Analog Gyro",
  "gyro ctor": "0",
  "is drivetrain": false,
  "motor controllers": [
    "VictorSPX",
    "VictorSPX"
  ],
  "number of samples per average": 1,
  "primary encoder inverted": false,
  "primary encoder ports": [
    8,
    9
  ],
  "primary motor ports": [
    11,
    12
  ],
  "primary motors inverted": [
    false,
    true
  ],
  "secondary encoder inverted": false,
  "secondary encoder ports": [
    2,
    3
  ],
  "secondary motor ports": [
    2,
    3
  ],
  "secondary motors inverted": [
    false,
    false
  ],
  "velocity measurement period": 100
}
```
Here's a screenshot of this from the program:
[image]

what should we do to solve this problem?

calcmogul, 2022-02-22:
What does the console say? The robot program was deployed, but it may be crashing on startup. Make sure you're using the latest SysId (2022.3.1) and your roboRIO has the latest image (2022 v4.0).

PeterJohnson, 2022-02-22:
In particular, go to Info | About in sysid to validate the version.  To upgrade the system tools like sysid it's necessary to download and run the WPILib installer.

max-niederman, 2022-02-23:
I'm also having this problem on v2022.3.1-gd2e04ac the latest commit on `main` as of writing.

max-niederman, 2022-02-25:
As it turns out, I had selected the wrong connection option for our NavX-MXP. Selecting the correct option fixed the issue.

closed 2022-03-07 by calcmogul