wpilibsuite/sysid #400: [Feature Request] Limit Switch Support To Sysid

mdurrani808, 2022-03-08:
When characterizing high inertia and powerful mechanisms (elevators and arms mainly), teams want to be very careful that their mechanisms do not break or get over driven. On a normal robot, this is done with limit switches. By being able to just tell sysid what DIO ports my limit switches were on and if they are normally true/false, it would be a lot easier to safely characterize my mechanisms. This feature could also be used in lieu of the disable button on certain mechanism.

Piphi5, 2022-03-08:
Would this be a single limit switch?

mdurrani808, 2022-03-09:
Two limit switches would make the most sense (mechanism up/down) but I assume most teams will just use 1.
And if soft limits could be integrated, that would be probably be best, accessibility wise.

calcmogul, 2023-09-21:
OBE by #518.

closed 2023-09-21 by calcmogul