wpilibsuite/sysid #402: CAN motor controllers report timeout/stale CAN messages

calcmogul, 2022-03-15:
We've been getting widespread reports of CAN motor controllers not moving after either SysId C++ binary is deployed. We've confirmed they're using the new roboRIO image, and the console prints messages about CAN message receive timeouts and the messages being stale. The robots under test generally only had like 4 motors installed, so CAN bus overload is unlikely. We've gotten a lot more reports with Falcons than with NEOs, but that could be due to relative marketshare.

3512 has gotten stale messages in non-SysId robot code too (C++ with RT thread) with Spark Maxes. I don't know if it was ever addressed.

At this point, I think the most likely cause is the vendordeps not playing nicely together since the SysId binary links them all in.

WPILib has no physical hardware, so we have no way to troubleshoot this.

calcmogul, 2022-03-15:
3512 has 9 NEOs, and they also make TimedRobot run at 5ms. This wasn't a problem in 2020/2021, but seems to be an issue for 2022.

modelmat, 2022-03-24:
Have confirmation from team 5032 [here](https://discord.com/channels/176186766946992128/528555967827148801/956699320210579547) that a [version of the code](https://cdn.discordapp.com/attachments/528555967827148801/956697266872279050/sysid-only-ctre.exe) with only the CTRE vendordep works.
Going to push the source for that .exe in just a second.

trevnels, 2022-05-25:
6502 may have seen similar behavior during this past season on Spark MAXes. If I remember correctly, we saw CAN timeouts after deploying sysid, but motors _would_ move when enabled - just in an erratic fashion (as if they were quickly starting and stopping.) Fully power cycling the robot after deploying sysid usually resolved it. 

Back then, my hunch was that there was something going wrong during the transition between previously deployed code and sysid which was fixed by a reboot, though I'm not familiar enough with sysid's internals to know if this theory makes sense.

calcmogul, 2023-09-21:
OBE by #518. This only ever occurred with the SysId embedded binaries, not user robot programs.

closed 2023-09-21 by calcmogul