wpilibsuite/sysid #406: Add CANivore support

calcmogul, 2022-03-19:
To support CANivore, we'd need to add an optional CANivore bus string field to the GUI for the CTRE devices that support it (not all of them do). An empty string would use the original constructor they use now.

ThadHouse, 2022-03-19:
Isnt there a sentinel value to use the roborio bus? So you wouldn't have to use an alternate constructor. And could just make the default in the UI be whatever that sentinel is.

Piphi5, 2022-03-20:
I think passing in `rio` defaults it to the roborio according to [here](https://docs.ctre-phoenix.com/en/stable/ch08a_BringUpCANivore.html#canivore-api).

Piphi5, 2022-03-20:
Since this will most likely be a 2023 change and will probably change the current hardware config json file format, would we want the currently saved hardware config files from this year to be compatible after the CANivore changes?

calcmogul, 2022-03-20:
SysId should be able to open old generator JSONs and autoconvert them to the new format, but the robot program doesn't need to be able to read the old format.

calcmogul, 2023-09-20:
Closed by #455.

closed 2023-09-20 by calcmogul