wpilibsuite/sysid #419: Improve robot-side logging to troubleshoot units and gear ratio problems

calcmogul, 2022-04-21:
From https://www.chiefdelphi.com/t/2020-2022-wpilib-feedback/409300/109

P33ge: "I had this issue as well. Was characterizing a drivetrain motor and was getting extremely inconsistent results."

calcmogul: "For feedforward or feedback?"

P33ge: "FeedForward. I posted my setup and no one found anything wrong. It was giving me 0.5 radians of rotation when it rotated several times. Encoder was setup correctly and everything. Ended up giving up because nobody found any issues."

calcmogul: "Well, it’s either a units issue or a gear ratio issue, which are the most common. Letting teams use their own robot program do the SysId logging wouldn’t address the problem either because they still need to figure out those two things."

calcmogul: "We could make the robot binary log more of the settings it loaded for troubleshooting."

Piphi5, 2022-04-22:
Would this just be adding more print statements on startup?

calcmogul, 2022-04-22:
Either that or NT logging.

Piphi5, 2022-04-22:
NT is probably more convenient (don't have to scroll through console) so we could just upload it to another entry

PR #421 (2022-12-11): Add CPR and Gearing reporting to NT
Resolves #419 

It reports gearing, cpr, and then conversion factor (gearing * cpr, except in the case of the spark max built-in or venom which are already in rotations.

Gearing and CPR are included since units are only referenced on the SysId side.
commit 146dccf0 (2022-12-11): Add CPR and Gearing reporting to NT (#421)

Fixes #419.
closed 2022-12-11 by calcmogul