wpilibsuite/sysid #446: Unable to fully lower dynamic step voltage

SusanC3, 2022-09-28:
Under voltage parameters, the slider only allows dynamic step voltage to be lowered to 2 volts. I'd like to use sysid to tune our robot's hood motor, but since the hood is fairly light and the acceptable range of motion is quite small, I'd never want to apply more than 0.5-1 volts. It would be nice if the step voltage could have smaller options, as well as being able to limit motion as described in #428.

calcmogul, 2022-09-28:
That lower limit is there to ensure the mechanism experiences a large enough acceleration. Having a mechanism that accelerates too fast with 1 V is very unusual; I'd consider adding a gear reduction if possible.

With that said, we could consider removing the limit if we also provide a metric for how well the dynamics have been sampled. We could probably do that by checking the singular values of X in the OLS equation y = Xβ somehow.

calcmogul, 2022-09-29:
Already fixed in https://github.com/wpilibsuite/sysid/pull/415.

closed 2022-09-29 by calcmogul