wpilibsuite/sysid #460: SysId NetworkTables json file not exist and crashed

shou692199, 2023-01-07:
**Describe the bug**
When using System Identification (SysId) tool. After deploying Drivertrain analysys type, robot code crashed with "networktable.json No such file or directory" error

**To Reproduce**
Steps to reproduce the behavior:
1. Open SysId
2. Set anything in Generator widget
3. Deploy to roboRIO

**Expected behavior**
Run logging tool normally

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - WPILib Version: 2023.1.1
 - OS: Windows 11
 - Java version: 17

calcmogul, 2023-01-07:
That's not what's causing the code to crash. There was a double-allocation when configuring Spark Maxes that causes the HAL to crash the program. It was fixed in https://github.com/wpilibsuite/sysid/pull/459, which didn't make it into the kickoff release.

You could recompile SysId yourself to get the fix or wait for the next release. We usually do them every week or so during build season.

sciencewhiz, 2023-01-17:
The updated SysID was included in the WPILib 2023.2.1 release

JaiCode08, 2023-01-17:
Upgrading WPILib 2023.2.1 did not work for me. I run and end that test and then the NT disconnects and connects over and over.

PeterJohnson, 2023-01-20:
Are you using the "Client (NT3)" or "Client (NT4)" mode when you're seeing the repeated disconnects?  Do you have any firewall/proxy software installed?

smoser-frc, 2023-01-21:
I'm seeing this also. We upgraded to 2023.2.1. We deploy fine.  Then we do not have Network Tables connectivity.

> Are you using the "Client (NT3)" or "Client (NT4)" mode when you're seeing the repeated disconnects? 

We have tried both NT3 and NT4.  When robot code is deployed (our base robot), then we get a green NT Connected message.  When we deploy the sysid program, it goes red, with occasional flickers to green.

> Do you have any firewall/proxy software installed? Is the DS console giving any messages (e.g. is the robot code restarting, is it showing multiple NT connect/disconnects, etc)?

We see nothing in driver station log window, and 'No robot code' as the status message. 

We *are* using SparkMAX and were attempting to run sysid with the drive train, but https://github.com/wpilibsuite/sysid/pull/459 is reported as fixed in 2023.2.1.  Also have a NavX gyro.

JaiCode08, 2023-01-21:
I am using two Sparkmaxs and  a NavX gyro. Make sure you have imaged the roboRIO to the latest version. When in SysID select NT4 in client before deploying and running the tests. This fixed my issue.

smoser, 2023-01-24:
> I am using two Sparkmaxs and a NavX gyro. Make sure you have imaged the roboRIO to the latest version. When in SysID select NT4 in client before deploying and running the tests. This fixed my problem.

Robo rio info is below from web dashboard.  WRT selecting nt4,  I’ll try again Saturday (maybe Thursday).  I know we tried with both 3 and 4, but not sure about having it selected when deploying (but we did try whole cycle of power off/deploy /run several times. 

Firmware Version: 6.0.0f1
Image Version: FRC_roboRIO_2023_v3.1

smoser, 2023-01-25:
What other information would be helpful for me to collect?  I'll have access to the robot again tomorrow and Saturday.
I can ssh to the roborio and get whatever logs or information  you're interested in, or video running of sysid. 

Please just let me know what might be helpful. (note that I (smoser) am the same human as 'smoser-frc' who commented above).

smoser, 2023-02-01:
Any thoughts here? Its possible this is user-error, as this is the first time our team has done it.  Thanks,

JaiCode08, 2023-02-01:
Sorry for replying late, it's also my first time using SysID. I am not an expert 😅

calcmogul, 2023-09-29:
OBE by #518. The problematic robot projects SysId deploys have been removed. NT4 seems to connect relatively fine to user programs.

closed 2023-09-29 by calcmogul