wpilibsuite/sysid #466: SysID Logger 2023 not connecting to network tables.

StarDust69, 2023-01-12:
The SysId 2023 does not allow the robot to connect over NetworkTables. Our team tries to characterize the robot with the the 2023 RoboRio image and WPILib software and this problem was introduced. With the 2022 version everything worked appropriately. 
[image]
The NT connection tag seems to connect for a few ms sporadically, but it fails. We followed the recommendation of rebooting our RoboRio and robot but it does not solve the problem. When we deploy **our** robot code, NT's for the Logger seem to connect, but of course, it lacks the robot code to perform the characterization. I hope an update can fix the issue. Thanks!

calcmogul, 2023-01-12:
Have you confirmed the robot code is running? I don't know what motor controller you're using, but there's a bug that causes a crash loop for Spark Maxes. You can download a fixed build from https://github.com/wpilibsuite/sysid/actions/workflows/build.yml?query=branch%3Amain.

A screenshot of your whole SysId window would help.

StarDust69, 2023-01-12:
That was it! Thanks! I encountered some issues in which the deployed auto code was not executing and received a message of wrong mecanism, but after a few resets I was able to run it. Thanks for your support.

closed 2023-01-12 by StarDust69