wpilibsuite/sysid #470: Successful deploy, DS shows "No robot code"

w8wjb, 2023-01-18:
Using SysId v2023.2.1.

Generator config:
- Analysis type: Drivetrain
- 2 SparkMax Brushless, R Inverted
- Using encoder ports, not inverted
- ADXRS450 Gyro, SPI CS0
- 1 counts/rev
- 10.71:1 Gearing

The SysId deploy succeeds, but the light on the Driver Station labeled "robot code" stays red and any attempts to enable the bot fail. If we deploy our own code to the same bot, the light is green and it works fine.

I tried logging into the RoboRIO directly and running `./frcUserProgram`; this is the output:

```
Killing previously running FRC program...

********** Robot program starting **********
Camera Server Library Not Found
NT: Listening on NT3 port 1735, NT4 port 5810
Initializing logger
Read JSON
Setup motors
Setup SPARK MAX (Brushless)
Setup SPARK MAX (Brushless)
Setup encoders
Setup SPARK MAX (Brushless) Encoder Port
Setup SPARK MAX (Brushless) Encoder Port
Setup gyro
Setup SPI (Onboard CS0)
Setup ADXRS450
NT: Got a NT4 connection from 10.49.67.184 port 64121
NT: CONNECTED NT4 client 'shuffleboard' (from 10.49.67.184:64121)
NT: Got a NT3 connection from 10.49.67.234 port 63678
NT: CONNECTED NT3 client 'NT3@10.49.67.234:63678' (from 10.49.67.234:63678)
NT: Got a NT3 connection from 10.49.67.234 port 63681
NT: CONNECTED NT3 client 'NT3@10.49.67.234:63681' (from 10.49.67.234:63681)
NT: NT3 socket error: socket is not connected
NT: DISCONNECTED NT3 client 'NT3@10.49.67.234:63681' (from 10.49.67.234:63681): remote end closed connection
Terminated
```

PeterJohnson, 2023-01-20:
What other dashboard type applications do you have running?  It looks like there's at least 2?  If it's sysid, is sysid configured to use "Client (NT4)" or "Client (NT3)"?  Try using "Client (NT4)" if it's configured for "Client (NT3)".

PeterJohnson, 2023-01-22:
This issue is probably due to the ADXRS450 bug we just fixed in wpilib.

w8wjb, 2023-01-22:
As far as dashboard apps, I am deploying the SysId robot code from a Mac and using the SysId desktop tool from there as well, but I have to run the Windows-only Driver Station on another machine. The Driver Station is also starting Shuffleboard automatically.

How would "ADXRS450 bug" affect DriveRobot startup?

ThadHouse, 2023-01-22:
Just constructing an adxrs450 object in c++ causes the code to enter an infinite spin loop, and never reach the point where code tells the ds there is actually code running. This bug doesn’t affect Java, but since the sysid robot program is built with c++, it is affected. The next patch release will fix the bug.

AlphaLQ, 2023-02-01:
Are there any plans or updates about the next bug fix?

adriancampos, 2023-02-01:
Are there any workarounds for now? Is there a way to build sysid-projects with a development build of allwpilib?

PeterJohnson, 2023-02-07:
Fixed in 2023.3.2 release.

closed 2023-02-07 by PeterJohnson