wpilibsuite/sysid #471: Getting "Watchdog not fed within ..." and "CanFrame too stale" issues when using Sysid

jasonli0616, 2023-01-21:
No issues occur at all when running our robot code, however these issues happen when we are using Sysid.

When running any of the Sysid tests, we enable, the robot does not move, and the errors appear upon disabling the robot (see log below).

The robot worked a couple of times (with Sysid), got some results, then just stopped working out of the blue (no changes on our part).

We are on WPILib 2023.2.1, and the 2023 version of NI FRC Game Tools (both latest as of today).

---

Config:

```json
{
  "counts per rotation": 4096.0,
  "encoder type": "CANCoder",
  "encoding": false,
  "gearing denominator": 1.0,
  "gearing numerator": 1.0,
  "gyro": "Pigeon2",
  "gyro ctor": "10",
  "is drivetrain": true,
  "motor controllers": [
    "TalonFX",
    "TalonFX",
    "TalonFX"
  ],
  "number of samples per average": 1,
  "primary encoder inverted": true,
  "primary encoder ports": [
    30,
    1
  ],
  "primary motor ports": [
    20,
    21,
    22
  ],
  "primary motors inverted": [
    true,
    true,
    true
  ],
  "secondary encoder inverted": false,
  "secondary encoder ports": [
    29,
    2
  ],
  "secondary motor ports": [
    23,
    24,
    25
  ],
  "secondary motors inverted": [
    false,
    false,
    false
  ],
  "velocity measurement period": 100
}
```

This is the error log:

```json
[
    {
        "messageType": 2,
        "timestamp": -0.000030517578125,
        "seqNumber": 414,
[... 421 lines cut ...]
    },
    {
        "timestamp": 23.27947998046875,
        "seqNumber": 480,
        "numOccur": 1,
        "errorCode": -3,
        "flags": 1,
        "details": "CTR: CAN frame not received/too-stale.",
        "location": "CANCoder 30 GetPosition",
        "callStack": "Stack Trace not received for performance reasons.",
        "messageType": 0
    },
    {
        "timestamp": 23.280181884765625,
        "seqNumber": 481,
        "numOccur": 1,
        "errorCode": -3,
        "flags": 1,
        "details": "CTR: CAN frame not received/too-stale.",
        "location": "CANCoder 29 GetPosition",
        "callStack": "Stack Trace not received for performance reasons.",
        "messageType": 0
    }
]
```

RandomPerson495721, 2023-01-24:
I think we had a similar issue, I don't know what the actual cause is, but it seems to be related with network tables. We were able to work around it by going to the logger, resetting the telemetry, changing the mode from Client (NT4) to disabled, apply, then change it back to NT4 and applying again.

jasonli0616, 2023-01-24:
@RandomPerson495721 Thanks for the workaround; we were also able to get it working creating a custom logger (as found on a Chief Delphi post: https://www.chiefdelphi.com/t/sysid-broken-on-2022-2-1/401875). Seems like some other teams on the FRC unofficial discord server were also facing the same issue.

calcmogul, 2023-09-21:
OBE by #518.

closed 2023-09-21 by calcmogul