wpilibsuite/sysid #479: Sysid not working with SparkMax

yotam201410, 2023-02-05:
we have been trying to use SYSID in order to get kp,ki,kd,ks,kv, and ka constants of our drivetrain with SYSID but we jeep getting 
```﻿﻿﻿﻿﻿﻿
Project failed: Hall sensore measurement period must be in the range of [8, 64] ﻿
﻿﻿﻿﻿﻿﻿﻿﻿Warning ﻿﻿ 0 ﻿﻿ [phoenix-diagnostics] Server shutdown cleanly. (dur:0) ﻿﻿  ﻿﻿﻿
﻿﻿﻿﻿﻿﻿﻿﻿Warning ﻿﻿ 0 ﻿﻿ [phoenix] Library shutdown cleanly ﻿﻿  ﻿﻿﻿
﻿﻿﻿﻿﻿﻿ pure virtual method called ﻿
﻿﻿﻿﻿﻿﻿ terminate called without an active exception
```
last year and this year the robot didnt move (the code got deployed and run) but last year we've managed to find a workaround using
[SysidJavaLogger](https://github.com/RegisJesuitRobotics/SysIdJavaLogger) for this year Sysid says that the data doesn't get to the NT Entry I checked it in the Sysid code and the entry hasn't changed.
if anyone has a solution it would be very appreciated things I have tried include:
downloading a new wpilib suite
downloading the latest build of Sysid
compiling Sysid
updating spark max to the latest image
the robot it selfs works with the regular code and so with the Sysid Code (V1.2) (there are no errors reported) but when enabling the autonomous the SparkMax does not show any voltage going to the motors. here are the IDs of the motors:
10 left forward
11 left Backward
12 right forward 
13 right backward
[https://www.chiefdelphi.com/uploads/default/original/3X/f/b/fb2df2cdafea3aba86e27954d39b45dd6e581449.jpeg](here is the image from V1.2)
I have played with the motor's ID in which slot are they but it hasn't worked.

sciencewhiz, 2023-02-05:
Fixed by #478

sciencewhiz, 2023-02-06:
What was the version you tested it on? There's been two releases in the last 2 days, so I want to make sure you tested the right version. The current latest is 2023.3.2.

yotam201410, 2023-02-06:
yes, i did

Alextopher, 2023-02-06:
v2023.3.2 we see the same error

calcmogul, 2023-02-06:
Could you try this build from #481?
https://github.com/wpilibsuite/sysid/actions/runs/4105458814

yotam201410, 2023-02-06:
ill try it tomorrow when ill be at the workshop

Lann4daze, 2023-02-06:
We just tried the latest build on wpilib 2023.3.2, and we're still getting that hall sensore measurement error, and occasionally an `Error in '/home/lvuser/frcUserProgram': free(): invalid next size (fast): 0x00ce4c88`

yotam201410, 2023-02-07:
> Could you try this build from #481? We have no hardware to test on. https://github.com/wpilibsuite/sysid/actions/runs/4105458814

still getting the same error,
here is the config [file](https://easyupload.io/allvfs)

calcmogul, 2023-02-07:
I think it was missing initialization code. Try the build from https://github.com/wpilibsuite/sysid/actions/runs/4116398469 when it's complete.

yotam201410, 2023-02-07:
> I think it was missing initialization code. Try the build from https://github.com/wpilibsuite/sysid/actions/runs/4116398469 when it's complete.

it fixed the issue thank you very much

closed 2023-02-07 by yotam201410