wpilibsuite/sysid #483: Suggestion - "CANivore Name" should be "CAN Bus Name"

agasser, 2023-02-06:
The field should be "Motor CAN Bus Name", not "Motor CANivore Name". If you have no CANivore the "CANivore Name" would be blank, not "rio".

The input box could also stand to be a little bigger, it only shows 5 characters.
[image]

calcmogul, 2023-02-06:
The docs say to use "rio" if you're not using a CANivore.
https://api.ctr-electronics.com/phoenix/release/java/com/ctre/phoenix/motorcontrol/can/WPI_TalonFX.html#%3Cinit%3E(int,java.lang.String)

agasser, 2023-02-06:
Yes. Sorry, I wasn't really clear. I was trying to say that if the field is labelled "CANivore Name", it doesn't make sense that I would put anything into the field if I have no CANivore - I have no CANivore name.
It does, however, make sense to enter "rio" if the field is labelled "CAN Bus Name".

calcmogul, 2023-02-06:
The maximum bus name length is 31 characters (plus a null terminator) based on what a CTRE employee implemented in the CANivore support PR. We'd have to figure out a way to make a scrollable input box.

agasser, 2023-02-06:
> The maximum bus name length is 31 characters plus a null terminator based on what a CTRE employee implemented in the CANivore support PR. We'd have to figure out a way to make a scrollable input box.

It does scroll now, based on cursor position in the box. Making the box 31 characters wide is probably too much, but 5 is too few. It could be 15 characters and fit pretty nicely on the screen.

calcmogul, 2023-02-06:
Another issue is now we're stuck with the old config name "canivore name" unless we have two conversion steps.

calcmogul, 2023-09-21:
OBE by #518.

closed 2023-09-21 by calcmogul