wpilibsuite/sysid #484: Sysid Values change when switching to and from Drivetrain sides

Danial-Sy, 2023-02-06:
My team and I recently ran a sysid test and retrieved our values. We observed an issue, a difference between two values when switching to and from them. To clarify, the values for a given side change. That is, for example, going from “combined” to “left” gives one set of values, and then clicking on “left” on the dropdown again shows a different set, even though those values should not change. The images attached below show the problem with all sides.

[image]
[image]
[image]
[image]
[image]
[image]

calcmogul, 2023-02-06:
The gains actually will change depending on which data you give it. "Combined" is usually the best one to use.

Shiv-Shah, 2023-02-07:
> This doesn't seem like a bug. The gains will vary depending on which datset you give it, and they don't appear to vary much in the screenshots. "Combined" is usually the best option to use because it uses all data available.

To rephrase what is happening here. Daniel pressed on combined once. Then Changed to Left, and then Left again. When changing lefts twice this error occurs. This shouldn't be a dataset issue as the dataset is the same.

calcmogul, 2023-02-07:
Could you upload your analysis JSON so the issue can be reproduced?

Shiv-Shah, 2023-02-07:
[SysId.zip](https://github.com/wpilibsuite/sysid/files/10670920/SysId.zip)

That zip file contains the JSON

calcmogul, 2023-02-09:
Looks like the numbers in the analysis pane aren't updating until the dropdown item is selected twice.

Shiv-Shah, 2023-02-11:
Which values do we use for this senario.

calcmogul, 2023-02-11:
Use combined.

Shiv-Shah, 2023-02-11:
which combined? And how is that different from left and right?

calcmogul, 2023-02-11:
The dataset dropdown you've been using in the analyzer pane has "Left", "Right", and "Combined" options for drivetrain data. Left uses the left side's data, Right uses the right side's data, and Combined uses both.

Shiv-Shah, 2023-02-11:
However if you look at the screen shots above and use our dataset. What you'll see are different values for when you press combined right right combined. Hence combined has 2 values. Which one do we use?

calcmogul, 2023-02-11:
Again, you have to press the button twice because it's doesn't switch after pressing just once.

Shiv-Shah, 2023-02-11:
ahh didn't see the part for the button twice thank you for the info. Any idea as to why this issue occured?

calcmogul, 2023-02-11:
From my earlier post:

> Perhaps it's because the analysis code occurs before the dropdown's polling code.

Shiv-Shah, 2023-02-11:
got it thanks!

PR #529 (2023-09-29): Fix analysis not updating after selecting dataset
Fixes #484.
closed 2023-09-29 by calcmogul