wpilibsuite/sysid #490: Invalid Spark Max encoder options

agasser, 2023-02-22:
When a _SPARK MAX (Brushless)_ motor is selected, the attached encoder selections do not make sense. There are two options for an attached encoder:
1) Encoder Port - When in brushless mode, the integrated hall effect encoder _must_ be plugged into the encoder port. The integrated hall-effect sensor cannot be inverted, but there is an option to invert it. In fact, the [documentation says](https://codedocs.revrobotics.com/cpp/classrev_1_1_relative_encoder.html#a5a96bc8f706ed26cc67320d847e21691):
> This will throw an error if the user tries to set inverted while the SparkMax is Brushless and using the hall effect sensor.
2) Data Port - When the data port is used, it can be a [duty-cycle absolute, relative, or analog encoder](https://www.chiefdelphi.com/t/rev-robotics-2022-23-product-releases-updates/417743/290?u=stablebiscuit). There is no way to select the encoder type. Additionally, the option to invert the encoder is missing.

rzblue, 2023-02-22:
With regards to the first part of 2, currently only quadrature encoders are supported on Spark Maxes through the data port in sysid. There should be a note that makes this clear, though.

agasser, 2023-02-22:
> With regards to the first part of 2, currently only quadrature encoders are supported on Spark Maxes through the data port in sysid. There should be a note that makes this clear, though.

That might explain why the robot reports "No Robot Code" and logs nothing after deploying with the _Data Port_ option when I have a REV through-bore encoder attached with an Absolute Encoder Adapter. I was left guessing because there is no Driver Station log message or feedback of any kind.

PR #491 (2023-02-24): Fix inversion checkbox showing up on built in Spark MAX and TalonFX encoder settings
Fixes #490 
Fixes #489 

Spark MAX Bug introduced in #385 due to option indices changing.
closed 2023-02-24 by calcmogul